ArduPilot/ardupilot #13900: Document remaining undocumented LogMessages

peterbarker, 2020-03-26:
## Feature request

We have infrastructure in place to allow the onboard log messages to be documented.  There are lots of messages yet to be documented; this is the list.  This list was generated by running 

```
pbarker@bluebottle:~/rc/ardupilot(master)$ ./Tools/autotest/autotest.py build.APMrover2 drive.APMrover2.LoggerDocumentation
```

... and snipping out the list of undocumented messages.

```
AUTOTEST: Undocumented message: ACC1
AUTOTEST: Undocumented message: ACC2
AUTOTEST: Undocumented message: ACC3
AUTOTEST: Undocumented message: ADSB
AUTOTEST: Undocumented message: AHR2
AUTOTEST: Undocumented message: ARM
AUTOTEST: Undocumented message: AROT
AUTOTEST: Undocumented message: ARSP
AUTOTEST: Undocumented message: ASM1
AUTOTEST: Undocumented message: ASM2
AUTOTEST: Undocumented message: ASP2
AUTOTEST: Undocumented message: ATDE
AUTOTEST: Undocumented message: ATUN
AUTOTEST: Undocumented message: BAR2
AUTOTEST: Undocumented message: BAR3
AUTOTEST: Undocumented message: BCL
AUTOTEST: Undocumented message: BCN
AUTOTEST: Undocumented message: CAM
AUTOTEST: Undocumented message: CESC
AUTOTEST: Undocumented message: CMD
AUTOTEST: Undocumented message: COFS
AUTOTEST: Undocumented message: CRSE
AUTOTEST: Undocumented message: CSRV
AUTOTEST: Undocumented message: CTRL
AUTOTEST: Undocumented message: DMS
AUTOTEST: Undocumented message: DSF
AUTOTEST: Undocumented message: DSTL
AUTOTEST: Undocumented message: ECYL
AUTOTEST: Undocumented message: EFI
AUTOTEST: Undocumented message: EFI2
AUTOTEST: Undocumented message: ERR
AUTOTEST: Undocumented message: ESC1
AUTOTEST: Undocumented message: ESC2
AUTOTEST: Undocumented message: ESC3
AUTOTEST: Undocumented message: ESC4
AUTOTEST: Undocumented message: ESC5
AUTOTEST: Undocumented message: ESC6
AUTOTEST: Undocumented message: ESC7
AUTOTEST: Undocumented message: ESC8
AUTOTEST: Undocumented message: EV
AUTOTEST: Undocumented message: FOLL
AUTOTEST: Undocumented message: FTN1
AUTOTEST: Undocumented message: FTN2
AUTOTEST: Undocumented message: GMB1
AUTOTEST: Undocumented message: GMB2
AUTOTEST: Undocumented message: GMB3
AUTOTEST: Undocumented message: GPA
AUTOTEST: Undocumented message: GPA2
AUTOTEST: Undocumented message: GPAB
AUTOTEST: Undocumented message: GPS2
AUTOTEST: Undocumented message: GPSB
AUTOTEST: Undocumented message: GRAW
AUTOTEST: Undocumented message: GRXH
AUTOTEST: Undocumented message: GRXS
AUTOTEST: Undocumented message: GUID
AUTOTEST: Undocumented message: GYR1
AUTOTEST: Undocumented message: GYR2
AUTOTEST: Undocumented message: GYR3
AUTOTEST: Undocumented message: HEAT
AUTOTEST: Undocumented message: ICMB
AUTOTEST: Undocumented message: IMT
AUTOTEST: Undocumented message: IMT2
AUTOTEST: Undocumented message: IMT3
AUTOTEST: Undocumented message: IMU
AUTOTEST: Undocumented message: IMU2
AUTOTEST: Undocumented message: IMU3
AUTOTEST: Undocumented message: IOMC
AUTOTEST: Undocumented message: ISBD
AUTOTEST: Undocumented message: ISBH
AUTOTEST: Undocumented message: LAND
AUTOTEST: Undocumented message: LGR
AUTOTEST: Undocumented message: MAV
AUTOTEST: Undocumented message: MAVC
AUTOTEST: Undocumented message: MMO
AUTOTEST: Undocumented message: MON
AUTOTEST: Undocumented message: MSG
AUTOTEST: Undocumented message: NKF0
AUTOTEST: Undocumented message: NKF1
AUTOTEST: Undocumented message: NKF2
AUTOTEST: Undocumented message: NKF3
AUTOTEST: Undocumented message: NKF4
AUTOTEST: Undocumented message: NKF5
AUTOTEST: Undocumented message: NKQ
AUTOTEST: Undocumented message: OABR
AUTOTEST: Undocumented message: OADJ
AUTOTEST: Undocumented message: OF
AUTOTEST: Undocumented message: ORGN
AUTOTEST: Undocumented message: PARM
AUTOTEST: Undocumented message: PM
AUTOTEST: Undocumented message: POS
AUTOTEST: Undocumented message: POWR
AUTOTEST: Undocumented message: PRTN
AUTOTEST: Undocumented message: PRX
AUTOTEST: Undocumented message: PSC
AUTOTEST: Undocumented message: RAD
AUTOTEST: Undocumented message: RALY
AUTOTEST: Undocumented message: RCDA
AUTOTEST: Undocumented message: RFND
AUTOTEST: Undocumented message: RPM
AUTOTEST: Undocumented message: RSSI
AUTOTEST: Undocumented message: SBFE
AUTOTEST: Undocumented message: SBPH
AUTOTEST: Undocumented message: SBRE
AUTOTEST: Undocumented message: SBRH
AUTOTEST: Undocumented message: SBRM
AUTOTEST: Undocumented message: SIM
AUTOTEST: Undocumented message: SITL
AUTOTEST: Undocumented message: SMOO
AUTOTEST: Undocumented message: SOAR
AUTOTEST: Undocumented message: STER
AUTOTEST: Undocumented message: STRT
AUTOTEST: Undocumented message: TEC2
AUTOTEST: Undocumented message: TECS
AUTOTEST: Undocumented message: TERR
AUTOTEST: Undocumented message: THR
AUTOTEST: Undocumented message: TRIG
AUTOTEST: Undocumented message: TSYN
AUTOTEST: Undocumented message: UBY1
AUTOTEST: Undocumented message: UBY2
AUTOTEST: Undocumented message: VAR
AUTOTEST: Undocumented message: VISO
AUTOTEST: Undocumented message: VISP
AUTOTEST: Undocumented message: WENC
AUTOTEST: Undocumented message: XKF0
AUTOTEST: Undocumented message: XKF1
AUTOTEST: Undocumented message: XKF2
AUTOTEST: Undocumented message: XKF3
AUTOTEST: Undocumented message: XKF4
AUTOTEST: Undocumented message: XKF5
AUTOTEST: Undocumented message: XKFD
AUTOTEST: Undocumented message: XKQ
AUTOTEST: Undocumented message: XKV1
AUTOTEST: Undocumented message: XKV2
```

**Describe the solution you'd like**
Crowd-sources documentation

**Describe alternatives you've considered**
Doing it myself.  Rather time consuming....

**Platform**
[ X ] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

peterbarker, 2020-03-27:
{code}
are there some instructions on how to add message documentation?

17:48
Nope.  Would you like some? :-)
1.  Find a message that needs commenting.
2. copy/paste an existing documentation block
3. Modify that block be doing sometimes serious amounts of code-delving
{code}

rishabsingh3003, 2020-03-27:
I'd like to try and help out with the plane documentation as a start!

peterbarker, 2020-03-27:
@rishabsingh3003 great!

b810a0de232680210fb2e3df250dc2839f2e32ea is a commit showing adding a single piece of documentation.

Running the autotest line in the first message in this PR (suitably adjusted for Plane) will validate the new message.

peterbarker, 2020-03-28:
Updated to reflect Plane messages as done (still in PR, 'though)

andyp1per, 2020-03-29:
@peterbarker https://github.com/ArduPilot/ardupilot/pull/13741/commits/8608f5d429fb46d6833d1a28b01584bb9d9f5a57

rishabsingh3003, 2020-04-22:
@peterbarker I am having a little trouble with Rover documentation. Is there any reason why [`DPTH`](https://github.com/ArduPilot/ardupilot/blob/cd6024741c563ba64c58ae0732f0cfa6ee651861/Rover/Log.cpp#L60) and [`SAIL`](https://github.com/ArduPilot/ardupilot/blob/cd6024741c563ba64c58ae0732f0cfa6ee651861/Rover/Log.cpp#L141) log messages not part of this list above.  

I am getting the following error when I am adding those two:
```
AUTOTEST: Failing tests:
  LoggerDocumentation (Test Onboard Logging Generation) (Documented message (DPTH) not in code) 
```

peterbarker, 2020-04-22:
@rishabsingh3003 parsing failure in my autotest code.  PR coming.

IamPete1, 2021-08-16:
Looks like this test is now running in CI so all messages are documented?