ArduPilot/ardupilot #13903: AP_VisualOdom: improve Intel Realsense T265 integration

rmackay9, 2020-03-27:
Here are a few things we could do to improve integration with the T265 discovered during my testing ([blog](https://discuss.ardupilot.org/t/easier-setup-for-intel-realsense-t265/53991))

- Move handling of the mavlink message to AP_VisualOdom ([issue](https://github.com/ArduPilot/ardupilot/issues/8061))
- Calibration to rotate the camera's frame (for attitude and position) into the AHRS/EFK's frame.  This includes rotating the camera frame to match the AHRS/EKF compass.
- Pre-arm check to ensure the camera's reported attitude is close to the AHRS attitude
- Handle camera reported resets of attitude or position
- Measure lag between camera and autopilot IMU and ensure EKF is use the correct lag

rmackay9, 2020-04-10:
PR https://github.com/ArduPilot/ardupilot/pull/13982 resolved 3 of the issue above

anbello, 2020-05-05:
Regarding first point under "Complete:"
"Measure lag between camera and autopilot IMU and ensure EKF is using the correct lag."

Where was this implemented? In the code? As parameter?
For EKF2 we have EK2_EXTNAV_DELAY param, but I don't see the analogous for EKF3. It should be useful as parameter to adjust it for different device or different method to send pose estimate.

rmackay9, 2020-05-05:
@anbello, Tridge has a script that can estimate the lag.  I'm not sure where it is but it came up with 15ms.

Thanks for the reminder on the lag for EKF3.  I've added an item into the list above.

60ericleal60, 2020-05-20:
I am unsure if this is the right place to discuss this because it may be ground station software related but I am having issues with viso offset values with the t265 camera and the APSync image. I have an airframe where the t265 camera and imu are pretty far apart (250mm) and the imu is mounted at the mass center which I select as the vehicle origin and place all of the other sensors offsets from this origin. If the vehicle is rolled right about its mass center viso y moves left and local position ned slowly moves to that position as if the viso offsets do not exist. As an experiment I set the camera position very above and below the imu (+/- 1.5m) with no change in result. After updating the firmware to today's file the viso position  parameter no longer shows up on the ground station even after updating to Mission Planner Beta. Is the implementation of offset and orientation being moved?

rmackay9, 2020-05-20:
@60ericleal60, The t265 support is our stable release is a bit weak so the best place to discuss may be on the developer gitter channel although some other developers may object: https://gitter.im/ArduPilot/VisionProjects

Are you sure that you've got Copter-4.0.4-dev on the board?  I suspect you've somehow installed Copter-4.0.3.

The test you're running is interesting, maybe you could provide a log file?

60ericleal60, 2020-05-20:
Here are some logs. 19 is an actual flight where the vehicle was erratic in position mode. I have a smaller airframe where the camera offset is small which flies very well. In 20 and 21 the propellers are off and I am rolling the vehicle about its mass center. In 20 there is a camera offset and in 21 there is not.
[viso_flights.zip](https://github.com/ArduPilot/ardupilot/files/4658222/viso_flights.zip)

60ericleal60, 2020-05-20:
When I update the firmware the viso offset parameters go away
[image]
[image]

rmackay9, 2020-05-21:
@60ericleal60,

Ah, you just need to set the VISO_TYPE to 2 and refresh the parameters.  This is a common thing we do in ArduPilot to hide parameters for users who aren't using the feature.

I've discussed with the other devs and you're more than welcome to join us on the gitter VisionProjects channel.  https://gitter.im/ArduPilot/VisionProjects

60ericleal60, 2020-05-21:
Thank you that worked and the controller now appears to be using those offsets. I then tried flying the vehicle and its behavior in hover in position hold was erratic. It looks as if the position controller isn't outputting anything but I am unsure. This is my first time flying 4.0.4. I think the first thing I will check is if it still does well in GPS aided position hold.
[00000022.zip](https://github.com/ArduPilot/ardupilot/files/4665229/00000022.zip)

rmackay9, 2020-05-22:
@60ericleal60, looks like there may be issues with the RC input.  Maybe you're using a joystick instead of a normal transmitter?
[image]

I also see that the EK2_ALT_SOURCE = 1 (for rangefinder).  If you're really flying indoors this may be OK but otherwise I would return it to zero.

I agree it's best to go back and test with regular GPS.  It's important that you've got a good solid working vehicle otherwise we spend our support time on debugging issues unrelated to the non-GPS vision feature.

60ericleal60, 2020-05-22:
I am reading an frsky radio with a microcontroller and then augmenting 2 of the channels for obstacle avoidance. I have flown successfully with this method on px4 for a long time and had relied on the pixhawk checking the first and last bytes of that packet for loss of link and corrupted packets. For now I will bypass this.

Because the t265 outputs a z position should I remove the lidar?

rmackay9, 2020-05-22:
@60ericleal60, I would leave the lidar on the vehicle for now.  It is a good extra source of altitude information and this is using for determining if the camera's scaling is correct (sometimes it's not and we are still investigating how to protect against this).  If the vehicle is being flow indoors on a vehicle with short legs sometimes the barometer altitude also can't be trusted so having a 3rd source is valuable.

60ericleal60, 2020-05-23:
Flight 0022 above is outdoors so there is so change in terrain. Is there an issue with the viso z position and range finder distance moving away from eachother?

Here are some flights on my smaller airframe indoors with a flat floor. It is on 4.0.3 and I suspect the sensor offset is small enough not to be catastrophic. I have set the position gains quite high here so I am not terribly concerned  with the large changes in attitude.

https://www.youtube.com/watch?v=a0IZEzS0GoI&feature=youtu.be
https://www.youtube.com/watch?v=zxfDPDcYQ74&feature=youtu.be

[mini_quad_viso.zip](https://github.com/ArduPilot/ardupilot/files/4671179/mini_quad_viso.zip)

60ericleal60, 2020-06-05:
After starting over on the vehicle setup with 4.0.4 and excluding the ground laser ranger, I am still unsure if the sensor offset values are being used. It appears like the offset compensation may occer on the raspberry pi from the mavlink message vision position delta but those values are 0 no matter what sensor offsets I put in. Do the vision sensor position offsets get calculated on the flight controller of the companion computer?

amilcarlucas, 2021-02-18:
I think https://github.com/ArduPilot/ardupilot/pull/16627 belongs in that list.

rmackay9, 2021-02-18:
@amilcarlucas thanks for bringing this up but I don't think it is directly related.  The T265 provides a local position estimate but we put this together with the EKF origin (lat, lon, alt) to determine the vehicle's global position.

Chobitsfan's PR adds support for missions to be specified in a local frame but this is useful regardless of whether the vehicle is using GPS, T265, optical flow or whatever else.

amilcarlucas, 2021-02-18:
My point here is that:

- T265 will be mostly used in indoor environments
- indoor environments as usually described in NED coordinates and not in lat, lon
- the setpoints should therefore be given in NED

But yes, this is not directly connected to T265, but it is still very relevant IMHO

shubham-shahh, 2022-02-14:
> My point here is that:
> 
> * T265 will be mostly used in indoor environments
> * indoor environments are usually described in NED [m] coordinates and not in lat, lon, alt
> * the user setpoints should therefore be given in NED
> * the copter will of course internally continue to use lat,lon ,alt
> 
> But yes, this is not directly connected to T265, but it is still very relevant IMHO

Hi, in my opinion, the T265 implementation with ardupilot is just an example of how external SLAM algorithms can be used with ardupilot, that said SLAM algorithms are also used for outdoor navigation, so implementing it as (indoor-only) is not a good thing in long run