ArduPilot/ardupilot #13922: Allow vectored VTOL yaw to be active during transition wait period

Hwurzburg, 2020-03-29:
Have yaw control active during transition wait period on vectored yaw QuadPlanes. Perhaps on non-vectored yaw ones also, but vectored yaw case would put much less demands on the overall thrust/current than normal quad/hex/octo configurations...

Currently, there is no yaw correction for disturbances in transition wait period since already in fixed-wing mode which has no yaw correction and are below flying speed for flying surfaces to be effective anyway...

Reason: long transitions in gusty conditions can cause significant yaw disturbance

**Platform**
[  ] All
[  ] AntennaTracker
[  ] Copter
[x] Plane
[  ] Rover
[  ] Submarine

IamPete1, 2021-08-16:
I wonder if this is fixed by the tiltrotor improvements we had over the last year.

Hwurzburg, 2021-08-19:
nope....tried in SITL on master....nada