ArduPilot/ardupilot #13932: POSITION_TARGET_LOCAL_NED doesn't correctly reflect SET_POSITION_TARGET_LOCAL_NED

hurowits, 2020-03-30:
## Bug report

**Issue details**

POSITION_TARGET_LOCAL_NED doesn't correctly reflect SET_POSITION_TARGET_LOCAL_NED. 
It always returns MAV_FRAME_LOCAL_NED, regardless of what was specified in SET. 
The type mask is also incorrect - it can only return 3 different masks, when there are many more possible combinations. 
Finally, yaw and yaw_rate are always zero. 

**Version**
4.03

**Platform**
[  ] All
[  ] AntennaTracker
[x] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

amilcarlucas, 2020-03-30:
[POSITION_TARGET_LOCAL_NED](https://mavlink.io/en/messages/common.html#POSITION_TARGET_LOCAL_NED) is the vehicle current position. It will almost never reflect [SET_POSITION_TARGET_LOCAL_NED](https://mavlink.io/en/messages/common.html#SET_POSITION_TARGET_LOCAL_NED), it is not meant to do that. The bug here is teh sentence 

> This should match the commands sent in SET_POSITION_TARGET_LOCAL_NED if the vehicle is being controlled this way.

in the descrition of the [POSITION_TARGET_LOCAL_NED](https://mavlink.io/en/messages/common.html#POSITION_TARGET_LOCAL_NED) message. That description is missleading. it should read instead:

> This should follow (with some lag) the commands sent in SET_POSITION_TARGET_LOCAL_NED if the vehicle is being controlled this way and the controll loop can keep up with the requested speed of position changes.

The other points you mention *are* missing features and bugs.

hendjoshsr71, 2021-10-31:
Most of this will be handled in #18580