ArduPilot/ardupilot #13946: Rover: Add support for TSYS01 driver

kklemens, 2020-03-31:
## Feature request

Customer request to add support for the [Celsius Fast-Response, ±0.1°C Temperature Sensor (I2C)](https://bluerobotics.com/store/sensors-sonars-cameras/sensors/celsius-sensor-r1/) for ArduRover.

Support for the Library and Driver were added in 2017, but only for ArduSub, no other firmware type. [AP_TemperatureSensor: Add AP_TemperatureSensor library and TSYS01 driver](https://github.com/ArduPilot/ardupilot/commit/ca603e151746d872b50aadc5f3a1bc0f7b6aaacd)

**Is your feature request related to a problem? Please describe.**
No technical issue other than this driver is missing from Rover.

**Describe the solution you'd like**
Please include support for this sensor in ArduRover. In QGroundControl, the readings show up under the "Temperature (3)" value.

**Describe alternatives you've considered**
According to the Peripheral Hardware section of ArduRover, there are no other external temperature sensor options other than the one for reading internal temperature of the autopilot.

**Platform**
[  ] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[X] Rover
[  ] Submarine

**Additional context**
Useful for obtaining surface water temperature readings.

thelowercaseguy, 2020-04-29:
@khancyr I think #14228 solves the issue. Let me know if there are any discrepancies :)

thelowercaseguy, 2020-04-30:
Hi, @khancyr @kklemens could you take a look at the PR #14239 I think this should solve the issue. Any disrepencies let me know  :)

kklemens, 2020-04-30:
I can test once its in master with one of the sensors, but I'm not super great at testing PRs. Thank you very much for doing this though! It will help with things in the future ;)

peterbarker, 2021-01-04:
@kklemens are you still interested in this functionality?

Sadly we do require code to be tested *before* it goes into master, rather than *after* :-)

I can build you binaries easily enough once we have a workable PR.

kklemens, 2021-01-05:
@peterbarker I haven't had many more customer requests, so low priority.

If you do get a binary together, I'd be happy to test it on a Pixhawk 1 and using one of these sensors. kevin@bluerobotics.com

rmackay9, 2022-02-10:
By the way, for boats, sonar often provide the water temperature and our NMEA 0183 driver supports this and logs it to the DPTH message.

hendjoshsr71, 2022-03-16:
Removed the good first issue label since I have a PR that shall eventually solve this one linked in above

ES-Alexander, 2025-11-21:
I believe this was technically completed in #21662, although it's not _practically_ solved because (as far as I can tell) only ArduSub is reporting temperatures in a reasonably device-independent way (at least using MAVLink - albeit somewhat awkwardly [shoehorned into the `SCALED_PRESSURE3` message](https://github.com/ArduPilot/ardupilot/blob/master/ArduSub/GCS_MAVLink_Sub.cpp#L119)).

Ideally [there would be a dedicated `TEMPERATURE` message in MAVLink](https://github.com/mavlink/mavlink/pull/2359#issuecomment-3442833067), so I suppose that upstream protocol change will need to be the next step...

peterbarker, 2026-11-15:
Closed by [PR #21662](https://github.com/ArduPilot/ardupilot/pull/21662). Since Rover 4.4 (1254e18fa9) the temperature sensor library is part of every vehicle, so Rover can use the TSYS01 with TEMP1_TYPE=1. It is compiled in by default on boards with more than 2 MB flash (a8ce43426b, [PR #28358](https://github.com/ArduPilot/ardupilot/pull/28358)), and available from the custom build server otherwise. Readings are logged as TEMP. Live GCS reporting needs a MAVLink TEMPERATURE message and should be tracked separately. Closing.

closed 2026-11-15 by peterbarker