ArduPilot/ardupilot #14005: CI GyroFFT test unreliable

peterbarker, 2020-04-06:
## Bug report

**Issue details**

Test fails randomly.

**Version**
master

**Platform**
[ X ] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
n/a

**Hardware type**
SITL

**Logs**

https://api.travis-ci.org/v3/job/671429105/log.txt

```
Failure Summary:
  test.CopterTests2:
    GyroFFT (Fly Gyro FFT) (Detected 178.720973Hz motor peak at -8.655870dB) (see /home/travis/build/ArduPilot/buildlogs/ArduCopter-GyroFFT.txt)
```

```
AUTOTEST: Exception caught: Traceback (most recent call last):
  File "/home/travis/build/ArduPilot/ardupilot/Tools/autotest/common.py", line 3281, in run_one_test
    test_function()
  File "/home/travis/build/ArduPilot/ardupilot/Tools/autotest/arducopter.py", line 3921, in fly_gyro_fft
    raise ex
NotAchievedException: Detected 178.720973Hz motor peak at -8.655870dB
```

IamPete1, 2022-09-11:
@peterbarker @andyp1per Any updates?

andyp1per, 2022-09-13:
I think it is what it is - probabilistic in its outcomes

closed 2022-09-13 by andyp1per