ArduPilot/ardupilot #14017: Create LUA binding for GCS heartbeat

cjmssmd, 2020-04-07:
## Feature request

I’d like ardurover to HOLD if no heartbeat has been received from the GCS in 1 second. I think the best way to implement this is with a LUA script that checks the age of the heartbeat from the GCS. But I don’t believe this binding is currently available… Correct? Next time you’re adding a new binding could you add this one??

https://discuss.ardupilot.org/t/lua-binding-to-gcs-heartbeat/54283

IamPete1, 2021-07-18:
Any reason why the existing GCS failsafe functionality does not work in your case? Scripting MAVLink is in the works, https://github.com/ArduPilot/ardupilot/pull/13660

IamPete1, 2024-05-04:
There are a few ways to do this now. You can either use the MAVLink bindings to read in the heartbeat directly. Or much simpler would be to use the `gcs:last_seen()` to get the time when the last message from the gcs was seen.

closed 2024-05-04 by IamPete1