ArduPilot/ardupilot #14024: Are the units for DO_MOUNT_CONTROL.pitch/roll/yaw in degrees?

DonLakeFlyer, 2020-04-08:
I'm trying to validate that QGS is working correctly with respect to ths command. Current QGC implementation has these values as degrees which matches mavlink spec. I asked if that matches ArduPilot and was told that the value was centi-degrees not degrees. I then was this with some new code which allows the user to change the gimbal angle from the ui and was getting inconsistent results from SITL testing and testing with my Solo running OpenSolo 4.0.

Testing with SITL master ArduCopter, MNT_TYPE set to 1/Servo:
* Create a mission which adjust the gimbal pitch using DO_MOUNT_CONTROL. QGC sets pitch value of -45 degrees.
* Run the mission and watch the values coming back from MOUNT_STATUS. They report pitch to be -4500 centi-degrees
* Which as far as I can tell indicates that the value of -45 degrees sent in the command was interpreted correctly

Tesing with my Solo running OpenSolo 4.0:
* Send a DO_MOUNT_CONTROL.pitch with -45 degrees value.
* Observe that the gimbal doesn't really move and that MOUNT_STATUS reports back -45 centi-degrees.
* Send a DO_MOUNT_CONTROL.pitch with -4500 centi-degrees value.
* Observe that the gimbal now pitchs down to 45 degrees and tMOUNT_STATUS reports back -4500 centi-degrees.

So what is going on here? Are the values supposed to he degrees or centi-degrees? I'm lost.

olliw42, 2020-04-08:
ArduPilot has a serious bug in this area, which I've reported several years back, but which just got ignored, and some Ardu devs in fact even simply claimed that ArduPilot has no such bug. So it's still there. A question you first want to figure out is what you want to listen to. If it should happen that you are genuinley interested in resolving the issue, you might want dig out my old report. Otherwise, my short answer to you would be that you are doing all correctly in QGC.
(and that I respond to you in this repo here I'd like to be seen as an act of unlimited kindness, it's sad to see you having to waste time for what is well known) 
good luck :)

peterbarker, 2020-04-08:
@DonLakeFlyer Yep, definitely looks like a problem.  Whether we can fix this without breaking existing Solos I'm not sure.

So reporting looks correct - i.e. the angles we're sending back look OK in each backend - they seem to be sending back cd.  I'm inferring from datatype in the storm32-serial driver; it's passing on values received from the gimbal in an int16_t, so it pretty much has to be cd.  All of the others are multiplying their targets while sending the message.

*Setting* angles is another matter.

Firstly I'd like to look at just handling direct mavlink traffic rather than missions.  Mission items get all sorts of munged for storage and whatnot.

I think we're in agreement that `DO_MOUNT_CONTROL` ought to be sending through degrees.

MAVProxy doesn't use `DO_MOUNT_CONROL`
MissionPlanner doesn't use `DO_MOUNT_CONTROL`
Artoo doesn't use `DO_MOUNT_CONTROL`; it uses the mavlink message `MOUNT_CONTROL`.  `MOUNT_CONTROL`'s fields are in centidegrees, and this is probably where the confusion (in ArduPilot) comes from.

Nobody appears to actually use `DO_MOUNT_CONROL`, which is exciting, and a touch sad....  So.  I wonder if we can actually *just fix* this without breaking anything?

I think the problem is simply https://github.com/ardupilot/ardupilot/blob/master/libraries/AP_Mount/AP_Mount.cpp#L583 - that should be multiplying the roll into centidegrees.  

Here's where we're handling `MOUNT_CONTROL` whose fields are centidegrees: https://github.com/ardupilot/ardupilot/blob/DEBUG%2Ff35-spi/libraries/AP_Mount/AP_Mount_Backend.cpp#L50

I think you would find that if you were to send the same commands you were sending to Solo via mavlink that it would behave the same - but it would be great if you could test that.

DonLakeFlyer, 2020-04-11:
> I think you would find that if you were to send the same commands you were sending to Solo via mavlink that it would behave the same - but it would be great if you could test that.

I think this is what you are asking for. I tested sending a direct DO_MOUNT_CONTROL command with a pitch of -45 degrees to SITL master copter. MOUNT_STATUS reported that back as -44 centi-degrees.

auturgy, 2020-04-12:
DO_MOUNT_CONTROL is implemented, at least for STorM32 : (https://github.com/ArduPilot/ardupilot/blob/ede87d49f68995b79e3b1aaebc4e3c0ced153731/libraries/AP_Mount/AP_Mount_SToRM32.cpp#L128
and appears to be implemented correctly.  Having the MSG and CMD with different units seems to be the problem (and doesn't make much sense).

rmackay9, 2022-08-26:
I think this issue has been fixed by PR https://github.com/ArduPilot/ardupilot/pull/21530 which will be included in 4.3.

Sorry it took so long!

closed 2022-08-26 by rmackay9