ArduPilot/ardupilot #14025: LED WS2812 setting with Mission Planner via MAVPIXEL

FrieKi, 2020-04-08:
## feature request

** Is your functional requirement related to a problem? Please describe. ** **.
To make the control of the LED WS2812 via Lua Script easier, see Betaflight / iNav / ... since not everyone masters programming via Lua.

** Describe the desired solution **
The insertion of a tool like MAVPIXEL or the like.

** Describe alternatives you have considered **
Converting a text file from MAVPIXEL to a LUA script.

**Platform**
[  ] All
[] AntennaTracker
[] Copter
[X] level
[] Rover
[] Submarine

** Additional context **

jdowdle, 2021-03-04:
I support this feature request.  I am trying to get the WS2812 leds on a Flywoo F745 nano board to operate with Arducopter.  They look really cool on Betaflight and Inav using the "Cylon" style built-in control.  All you have to do is enable and setup in this screen.  I have spent hours on Arducopter, and have gotten a build to otherwise work on the board, but cannot succeed on the leds.  Hate to give them up.  I have followed all documentation, and searched for solutions.  Trial and Error.....  Arducopter is otherwise much preferred over the other autopilots in most regards.

FlyingRobotZA, 2023-05-22:
I also support this feature.  Really easy to use on Betaflight and iNav.  Not really experienced with Lua Scripts so now have to try find someone and pay them to do a simple LED layout.

andyp1per, 2023-05-24:
Can you provide additional context on what MAVPIXEL does?

FlyingRobotZA, 2023-05-24:
Hi @andyp1per You can view the Git here https://github.com/prickle/Mavpixel  To see it in action in iNav you can see in this video https://youtu.be/kjHruLW8KT0

davidbuzz, 2023-05-25:
also ... 
https://github.com/prickle/MavpixelGUI

davidbuzz, 2023-05-25:
So, i think the original request can be described as :
 - allow ardupilot's neopixel implementation [ which is currently controllable by lua ] to be able to configured using the mavpixel GUI tool [ie over mavlink] without requiring dedicated 'mavpixel' hardware[which is basically a mavlink-mavpixel-to-neopixel converter]
I don't know if this could entirely be done as a LUA script or not, but it might be possible.[lua recieves mavpixel packets, and responds to them]

SteveintheIoW, 2025-04-09:
As others have mentioned, I am struggling to implement a desired WS2812 LED arrangement in AP LUA script. It is proving to be the most difficult part of my Ardupilot setup compared to INAV.  INAV LEDs were wonderfully simple to setup.... I use a strip of 3 neopixel LEDS folded into a U shape around a buzzer, Left side shows RED, Lower side is a White Strobe, Right side is Green (small, simple, lightweight on a sub250g quad)....Two weeks later i'm still struggling to implement it in AP!   Ardupilot is a far superior Autopilot in all other respects, a very impressive job everyone!