ArduPilot/ardupilot #14029: autotest: fails ThrottleFailsafe test (Radio failsafe SmartRTL->RTL)

tatsuy, 2020-04-09:
I feel like using the same code in Travis CI build sometimes fails and sometimes succeeds.

This is the evidence.
https://travis-ci.org/github/ArduPilot/ardupilot/jobs/672369580
https://travis-ci.org/github/ArduPilot/ardupilot/jobs/672752835
I rebased to run AutoTest again, but I think it will not affect the test result because it is "Tools" only change.

-- failed pattern --
22383param set SIM_GPS_DISABLE 1                          // GPS disable
22389APM: NavEKF2: lane switch 1
22390APM: EKF primary changed:1
22391APM: EKF variance
22392APM: EKF2 IMU0 has stopped aiding                // AID_NONE
22393APM: EKF2 IMU1 has stopped aiding
22394APM: SmartRTL deactivated: bad position         // cannot use Smart RTL
22402param set SIM_GPS_DISABLE 0                          // GPS enable
22409AHRS_EKF_TYPE = 2.0000000
22410AUTOTEST: Waiting for EKF value 831
22411AUTOTEST: Wait EKF.flags: required:831 current:831 errors=0
22412AUTOTEST: EKF Flags OK                                   // complete self.wait_ekf_happy()
22413AUTOTEST: Delaying 5.000000 seconds            // self.delay_sim_time(5)
22414APM: EKF2 IMU0 is using GPS                           // AID_ABSOLUTE
22415APM: EKF2 IMU1 is using GPS
22423param set SIM_RC_FAIL 1                                  // RC Disable
22426APM: Radio Failsafe
22427APM: Mode change failed: SMARTRTL requires position  // failed to change to Smart RTL
22428APM: SmartRTL Unavailable, Trying RTL Mode
22429APM: Mode change failed: RTL requires position              // failed to change to RTL
22430LAND> Mode LAND                                          // change to LAND
22431SIM_RC_FAIL = 1.0000000
22432AUTOTEST: Waiting for mode RTL                     // already set to LAND mode, so this test fails
----

-- success pattern --
param set SIM_GPS_DISABLE 1
APM: EKF variance
APM: EKF2 IMU0 has stopped aiding                        // AID_NONE
APM: EKF2 IMU1 has stopped aiding
APM: SmartRTL deactivated: bad position
AUTOTEST: Setting SIM_GPS_DISABLE to 0.000000
param set SIM_GPS_DISABLE 0                                  // GPS enable
APM: EKF2 IMU0 is using GPS                                   // AID_ABSOLUTE
APM: EKF2 IMU1 is using GPS
AHRS_EKF_TYPE = 2.0000000
AUTOTEST: Waiting for EKF value 831
AUTOTEST: Wait EKF.flags: required:831 current:831 errors=0
AUTOTEST: EKF Flags OK
AUTOTEST: Delaying 5.000000 seconds
AUTOTEST: Setting SIM_RC_FAIL to 1.000000
param set SIM_RC_FAIL 1                                           // RC Disable
APM: Radio Failsafe
APM: Flight mode change failed
APM: SmartRTL Unavailable, Trying RTL Mode
RTL> Mode RTL                                                          // success to change to RTL
AUTOTEST: Waiting for mode RTL
-----

I checked the differences and came to the following conclusions.
EKF's AidingMode has a delay in returning from AID_NONE to AID_ABSOLUTE.
In failed pattern, SIM_RC_FAIL is set immediately after switching to AID_ABSOLUTE.
So, the EKF position has not yet fully recovered and failed to switch to RTL.

Quick fix is to wait more than 5 seconds.
https://github.com/ArduPilot/ardupilot/blob/master/Tools/autotest/arducopter.py#L600

I wonder if it took time in returning from AID_NONE to AID_ABSOLUTE.
There could be a problem with EKF AidingMode Switching.

**Platform**
[  ] All
[  ] AntennaTracker
[〇] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

tatsuy, 2020-05-01:
this is another case (#14244) with the same issue.
 https://travis-ci.org/github/ArduPilot/ardupilot/jobs/681934852#L22341

```
APM: Radio Failsafe
APM: Mode change failed: SMARTRTL requires position
APM: SmartRTL Unavailable, Trying RTL Mode
APM: Mode change failed: RTL requires position
LAND> Mode LAND
SIM_RC_FAIL = 1.0000000
AUTOTEST: Waiting for mode RTL
```

peterbarker, 2020-05-02:
This is a case of the autotest suite being told by the autopilot that absolute position is available - but when we try to use that fact we get the "requires position" thing.  It manifests itself in several tests in different ways.

I can't see how it can be anything but a bug - we're announcing something which is just not true.

tatsuy, 2020-05-02:
Do you mean that the problem is only reported in autotest?
I hope it doesn't happen on the real vehicle.

peterbarker, 2020-05-02:
On Fri, 1 May 2020, Tatsuya Yamaguchi wrote:

> Do you mean that the problem is only reported in autotest?
> I hope it doesn't happen on the real vehicle.

I don't see any reason this can't happen on a real vehicle.

tatsuy, 2020-05-05:
ok, thank you for your comment.
I'll leave it open this issue in case someone else encounters the same problem.

Pedals2Paddles, 2020-05-05:
Closing this since it's a bug in the EKF described in this report https://github.com/ArduPilot/ardupilot/issues/13801.  Autotest is working correctly so no sense in having an autotest bug open on the same problem.

closed 2020-05-05 by Pedals2Paddles

tatsuy, 2020-05-06:
@Pedals2Paddles 
Sorry, I couldn't find the issue.
Thank you for telling me that.