ArduPilot/ardupilot #14031: Copter: Loiter Turns does not point vehicle towards center

rmackay9, 2020-04-09:
Unlike the Circle flight mode, Auto's Loiter-Turns command with lat and lon position specified does not point the vehicle nose towards the center of the circle.  It should and I believe it once did but it seems a bug has been introduced that has broken the behaviour.

To recreate the problem in SITL do this:

- rm eeprom.bin to restore all parameters to their defaults
- create a mission with a takeoff, LoiterTurns and RTL commands
- run the mission, then vehicle will not point towards the center of the circle, instead it's heading will remain at whatever it was when it reached the edge of the circle.

[image]

rishabsingh3003, 2020-04-09:
The issue might be somewhere else, but I believe this was the commit following which we have this bug:
https://github.com/ArduPilot/ardupilot/pull/5518/commits/bf7222dea52c581f1e9e6f82d0d5e75fa0fc33b9

rmackay9, 2020-04-21:
Resolved by the linked PR, thanks!

closed 2020-04-21 by rmackay9