ArduPilot/ardupilot #14034: GCS: Statustexts can consume all bandwidth space

WickedShell, 2020-04-09:
## Bug report

**Issue details**

Statustexts are opportunistically inserted into the outgoing MAVLink buffer if there is space available when they are emitted if they are being added from the main thread. The problem with this is that main loop thread can keep sending more and more status texts until the entire buffer has been consumed. It's a rare occurrence, as we try to avoid emitting it, but I have an actual log from a user in the field using a telemetry radio at 57600 where this happened to them due to repeated spamming of "EKF yaw reset" which got to be so excessive and fast that all downlinked telemetry except for the occasional heartbeat were lost. This condition persisted until poweroff, which if operating at long range would be extremely problematic as you would have no situational awareness of what the vehicle was doing. The tlog is not a log I have the rights to reproduce here.

To solve this we either need a general form of rate limiting on the status text being directly inserted, or just never directly emit the statustext, and instead always just service the queue from the stream rate system. It's nice to get the text into the outgoing queue as early as possible, but much like how the FTP and log download work, unless it participates in the streaming system it will always end up screwing up the normal stream, so I'm currently inclined to think we should be servicing it as part of the streaming (maybe with a check that says if we have sent everything normally scheduled *and* we still have buffer space then we could send out more status texts?) Open to other ideas on how to resolve this though.

**Version**
All stable releases, and master

**Platform**
[ x ] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

peterbarker, 2020-04-09:
Send the text immediately if we haven't sent one recently.  "Recently" is worked out from streamrate.

peterbarker, 2024-07-31:
Everything now goes via the queue, so it's all rate-limited.  Closing this one now.

closed 2024-07-31 by peterbarker