ArduPilot/ardupilot #14035: enable gzip compression for manifest.json

booo, 2020-04-09:
Currently the https://firmware.ardupilot.org/manifest.json file gets served without gzip compression. Please enable gzip compression for the file on your webserver.

Is this the right place to ask for such a feature?

peterbarker, 2020-04-09:
I've looked at doing the accept-transfer-encoding dance, and it would be a bit of a pain to set up on the server.  Perhaps @jpkh would like to play with it.  Definitely would want to NOT compress it every time it is requested

You can already request a pre-compressed version of the file - just tack on a `.gz` ;-)

I would like to know, however - for your particular use case would an `xz` be just as useful as a `gz`?  The size savings are notable.

https://github.com/ardupilot/ardupilot/blob/master/Tools/scripts/build_binaries.py#L691

booo, 2020-04-09:
Serving a pre-compressed file with nginx doesn't look too complicated. Apache seems to be a bit more complicated :)

Are you sure the result of the compression doesn't get cached? Maybe one could enable compression and caching on the server side to circumvent the compression on every request?

I think the webserver should honour requests for compression in the http header. I don't care about the particular compression algorithm.

tridge, 2020-04-09:
just request the file with .gz suffix. That is why it is there

WickedShell, 2020-04-09:
@tridge I'm sorry but that's not really the right answer. If that's the approach then the `.json` shouldn't be served. The point of letting the server negotiate it is most featured HTTP clients will be able to negotiate the compressed stream automatically, but if we rely on someone manually extracting the file then lazy programmers will skip that step sometimes. There are major GCS's which are skipping out on this. We are paying a server cost for other people requesting the wrong file. If we support compression on the stream then more people will be using a better format because their file fetching library can automatically manage it for them, and it's less load on our services.

booo, 2021-05-31:
Should I close the issue or can we make any progress on this in the next few weeks?

peterbarker, 2021-06-01:
@booo there's not a great deal of enthusiasm to chase it down.  As mentioned, I did stare at it for a while but the general honour-the-header stuff is done with on-the-fly-compression.  My gut feeling was that we should serve from the .gz file instead.  Perhaps I should just revisit that assumption - perhaps the fetching of manifest.json is infrequent enough that we should just compress it on the fly every time, at least until someone does it better.

It would be a good improvement, but it has to be done very carefully as we really don't want to break firmware downloads accidentally while we're doing it.