ArduPilot/ardupilot #14042: OSD Based Menu System

SpaceGoast, 2020-04-09:
## Feature request

**Is your feature request related to a problem? Please describe.**
I think ardupilot would attract many more users if it could offer an OSD menu system for tuning common parameters, instead of having to bring a laptop/ground station. I also think an OSD menu would facilitate faster new model setup. I have used iNav/Betaflight/Ardupilot and I am 10 times more likely to tweak PIDs using the OSD menu than to drag out my laptop. 

I would like to work on/solicit help building an OSD menu system for ardupilot. I am not a software developer, but have done some software projects here and there. I am happy to help document/architect/code but will likely need some hand holding to code.  

**Describe the solution you'd like**
I think the logical way to implement this would be stick gestures, but I have read a couple times there are safety concerns in the ardupilot dev community with stick gestures since ardupilot uses throttle/yaw to arm. I've thought about just using dedicated RC channels to enable this menu but 90% of the time it would be wasted channels(you couldn't use it if the vehicle was armed). 

My thought would be in the arming check file, add a parameter like "menu in use" so arming checks would fail if a menu is active, and even better make it so arming requests aren't even accepted while in a menu. To enter the menu there just needs to be a mutually exclusive window for throttle arming and entering a menu. i.e. throttle 0-15% for arm, throttle 40-60% for entering menu. If you require the throttle to always be above the arming window while menus are active, I don't see any reason why this couldn't be implemented in a safe manner. Of course osd menus could be disabled permanently by anyone who didn't want them active and I am open to other ideas of how to implement a menu system besides stick gestures. They are just intuitive and use existing channels instead of requiring a bunch of new channels.  There is the issue on plane where you couldn't enter a menu after arming since there is no disarm but that's minor. If I were to start with stick gestures, is that doomed to never be implemented into master? 

Things I would definitely want in a menu system:
-PID values (ability to tune common values)
-flight modes
-feature settings (smart audio vtx control in particular)

**Describe alternatives you've considered**
iNav

**Platform**
[X] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Additional context**
List of topics/requests for this kind of thing:
https://discuss.ardupilot.org/t/is-there-a-way-to-get-smartaudio-working/29025/39
https://github.com/ArduPilot/ardupilot/issues/10084
https://github.com/ArduPilot/ardupilot/issues/6680

andyp1per, 2020-04-09:
@SpaceGoast there is actually precedence for this in ArduPilot since my RunCam PR was merged. This add sticks gestures for RunCam OSD suppot - so I don't think there is any issue having more of these. In fact with that support arming is disabled while in the OSD menu so it's totally safe. More the issue is adding on-screen support is a reasonable chunk of work so you would need to find someone willing to do this. I do plan to work on CSRX support at some point (which is not far off smart audio) which will probably involve generalizing the stick control I have already done. Another option is to leverage the new scripting support.

peterbarker, 2020-11-25:
This has now been done; you can set parameters with your OSD.

https://ardupilot.org/copter/docs/common-paramosd.html

closed 2020-11-25 by peterbarker