ArduPilot/ardupilot #14046: AP_NavEKF2: writeExtNavData's resetTime_ms has no effect

rmackay9, 2020-04-10:
While testing the T265 camera support I found that updating the writeExtNavData() calls resetTime_ms argument has no effect on how the EKF consumes the external position data.  I would have expected a jump in the EKF's reported position if the external position source simultaneously provided a jump in position and updated the resetTime_ms argument but it didn't.  The EKF instead drifted the position to the new location.
[image]

From inspecting the code it doesn't seem that the extNavLastPosResetTime_ms nor the extNavDataNew.posReset are actually used.

I'm hoping that @priseborough can provide some insight and suggestions on how to fix this.

rmackay9, 2020-04-11:
I've created a branch here which makes it easier to reproduce the issue: https://github.com/rmackay9/rmackay9-ardupilot/commits/t265-scale

To reproduce in SITL do the following:

- cd ArduCopter
- rm eeprom.bin
- ../Tools/autotest/sim_vehicle.py --map --console
- param load ekf-ext-pos-reset.parm (this sets EK2_GPS_TYPE = 3, VISO_TYPE = 1)
- restart SITL again
- set the EKF origin to where the copter appears on the map (if using mavproxy right-mouse-button-click and select "Set Origin (with height)"
- you should see the vehicle slowly travelling North but every 30 seconds it will speed forward about 10m.  These occur just after the external position is moved forward 10m but the vehicle should really "jump" forward instead of "speeding" forward.

priseborough, 2020-04-11:
Looking at the code, below this line:

    extNavDataToFuse = storedExtNav.recall(extNavDataDelayed, imuDataDelayed.time_ms);

 I would  add the following to force a reset on that frame 

    // Check for and apply external reset
    if (extNavDataToFuse && extNavDataDelayed.posReset) {
        // apply and record horizontal reset
        posResetNE.x = stateStruct.position.x;
        posResetNE.y = stateStruct.position.y;
        stateStruct.position.x = extNavDataDelayed.pos.x;
        stateStruct.position.y = extNavDataDelayed.pos.y;
        outputDataNew.position.x = stateStruct.position.x;
        outputDataNew.position.y = stateStruct.position.y;
        outputDataDelayed.position.x = stateStruct.position.x;
        outputDataDelayed.position.y = stateStruct.position.y;
        posResetNE.x = stateStruct.position.x - posResetNE.x;
        posResetNE.y = stateStruct.position.y - posResetNE.y;
        lastPosReset_ms = imuSampleTime_ms;

        if (activeHgtSource == HGT_SOURCE_EV) {
            // apply and record vertical reset
            posResetD = stateStruct.position.z;
            stateStruct.position.z = extNavDataDelayed.pos.z;
            outputDataNew.position.z = stateStruct.position.z;
            outputDataDelayed.position.z = stateStruct.position.z;
            posResetD = stateStruct.position.z - posResetD;
            lastPosResetD_ms = imuSampleTime_ms;
        }
    }

rmackay9, 2020-04-11:
@priseborough,

Thanks for that, it's much better although it still seems to overshoot for a moment and then return.  As if perhaps the velocity or acceleration estimate is still consuming the position jump.  I'm not sure, it's just a guess.

Here is a picture of the latitude and raw vision sensor positions (i.e. the vision sensor positions are what is being fed into the EKF).

[image]

priseborough, 2020-04-11:
Plot the NKF1.PN and NKF1.PE (EKF internal states). if these are OK, thenwe need to add some code to increment the output observer state history buffer by the amount of the reset.

rmackay9, 2020-04-11:
@priseborough,

[Here's a link to the log](https://www.dropbox.com/s/w4sfnr1wl5akg0p/reset-fix.bin?dl=0) and below is a screenshot of the NKF1.PN, PE and VISO.ResetCnt.  It looks to me like the little overshoot is visible here too.
[image]

So maybe the issue is with how I'm testing.  The vehicle is not actually moving in SITL.  I'm just sending in a constant stream of vision-position-estimate messages that make it appear like it's moving at about 1m/s.  Then every 30sec I send it a big jump of 10m and set the reset counter at the same time.  So from the EKF's point of view it will see a mismatch between the accelerometers and the positions I guess.  Maybe this is the cause.

I'll make a better test on Monday in which the vehicle is actually moving and see if it acts any differently.

rmackay9, 2020-04-11:
I decided to do the test now instead of Monday.  So what I did was modify the AP_VisualOdom library so that it injected an additional 10m jump in the x-axis every 10seconds.  The results are pretty much the same in that we see an overshoot in the moments after the reset.

In the screeshot below the POS.lat (the EKF's estimated latitude) is shown in red (scale on the left) and the raw vision position is shown in blue (scale on the right).
[image]
[Here's the SITL log in case it's useful](https://www.dropbox.com/s/diiz6j78sshqyc5/vision-position-with-10m-resets.BIN?dl=0).

So I think you're right that we actually need to go back and modify the historical buffer of positions as well by the reset amount.

rmackay9, 2020-05-01:
This is fixed in master so closing, thanks!

closed 2020-05-01 by rmackay9