ArduPilot/ardupilot #14054: Copter: watchdog reset does not appear to create WDOG message

rmackay9, 2020-04-11:
The [watchdog feature](https://ardupilot.org/copter/docs/common-watchdog.html) does not appear to be creating the WDOG log file messages anymore.

I had a watchdog reset on a CubeOrange after removing the USB cable (while a battery was plugged in).  These messages appeared on the messages tab in Mission Planner but there were not WDOG messages in the logs on the SD Card. We checked all 150 logs on the SD card and none had the message.

2020-04-09 4:23:24 PM : IE0 IEC0
2020-04-09 4:23:24 PM : WDG: T-3 SL0 FL0 FT0 FA0 FTP0 FLR0 FICSR0 MM0 MC0 
2020-04-09 4:23:24 PM : Frame: QUAD
2020-04-09 4:23:24 PM : RCOut: PWM:1-12
2020-04-09 4:23:24 PM : CubeOrange 001D0023 34385113 39373231
2020-04-09 4:23:24 PM : ChibiOS: d4fce84e
2020-04-09 4:23:24 PM : ArduCopter V4.0.4-dev (73a0e8fe)

peterbarker, 2020-04-11:
@rmackay9 please verify that when you receive the watchdog reset that logging is *not* being force-started?

peterbarker, 2020-04-11:
I think this is actually a difference between Plane and every other vehicle, as Plane arms itself:

```
    if (hal.util->was_watchdog_armed()) {
        gcs().send_text(MAV_SEVERITY_INFO, "Forcing logging for watchdog reset");
        _params.log_disarmed.set(1);
    }
```

Plane is the only vehicle which will arm itself after reset.

Changing that line to `was_watchdog_reset` will probably fix the problem - and make it inline with the statustext :-)

rmackay9, 2020-05-12:
This is not resolved in master so closing

closed 2020-05-12 by rmackay9