ArduPilot/ardupilot #14055: GCS_MAVLink: send-text messages from peripherals not sent to GCS

rmackay9, 2020-04-11:
While working on the T265 I've found that send-text messages from mavlink enabled peripherals are not passed through the autopilot to the ground station.  This makes it difficult for these "smart" peripherals to keep the user informed as to their status especially during startup.

This issue has come up with two unrelated peripherals:

- The [Intel RealSense T265 scripts ](https://ardupilot.org/copter/docs/common-vio-tracking-camera.html) from Thien
- The [DrotagX geotagger](https://ardupilot.org/copter/docs/common-geotagging-drotagx.html)

peterbarker, 2020-04-11:
On Fri, 10 Apr 2020, Randy Mackay wrote:

> While working on the T265 I've found that send-text messages from mavlink enabled peripherals are not passed through the autopilot to the ground station.
> This makes it difficult for these "smart" [...]

This has come up somewhere before, relatively recently.

statustext doesn't have any target system id or component id.  That makes 
it a broadcast message.

That means mavlink routing can't work - if we choose to forward the 
message then we can't *not* forward it on a link based on having seen a 
target systemid/compid on that link.  This is the "process_locally" bit in 
MAVLink_routing; it's "for us", so we don't send it on.

If we were to forward it on a link you could relatively easily end up with 
a routing loop.  Consider several vehicles all on the same network segment 
using multicast, for example.  Consider what happens in a multi-link 
environment if a GCS (on a companion computer, for example) were to 
do the same thing and pass on broadcast messages.

What we may need is a "forward broadcast messages" flag on a mavlink 
channel, which does forward broadcast packets from a channel, which you 
could enable on uarts where required.  This might need to be "proxy 
broadcast messages" to rewrite the src and dest IDs, I haven't thought 
that far.

olliw42, 2020-04-11:
this is not correct. Broadcast messages including status text are correctly forwarded by ArduPilot. I know that for sure since I see that every day working with STorM32 and other MAVlink peripherals of mine. (in MP the status text of the component is displayed with the component ID in front). A peripheral sending MAVlink messages should however obey to some basic MAVLink standards. Please first analyze the situation carefully and correctly before jumping into adding strange complicated and unneeded mechanisms. I'll write this to save you your time.

auturgy, 2020-04-11:
Thanks Olli.
This isn’t a new issue for Dronekit iirc. I think the previous workarounds involved running telemetry from the companion computer not the autopilot (ie using a router, or a second connection to DroneKit).
I’d also remind people that for some reason DroneKit uses comp ID of 0 which isn’t a valid source ID.
If we’re going to continue to use/promote DroneKit, it needs to be maintained.

peterbarker, 2020-04-11:
Link to previous discussion: https://discuss.ardupilot.org/t/forward-statustext-to-gcs/42927/6

rmackay9, 2020-04-13:
I think @olliw42 is correct the send-text messages are getting through.  Below is a screen shot of the T265 getting it's message through to the ground station even though it's on the "far side" of the autopilot.
[image]
Sorry for the false alarm and thanks to @olliw42 for the correction.

closed 2020-04-13 by rmackay9