ArduPilot/ardupilot #14061: COPTER 4.0.3 Add failsafe when rc_override latency goes above X ms

roque-canales, 2020-04-11:
Hello,

We experience random lag issue when using RC override through 4G/3G connexion.

It's due to random lag in the 4G connexion, but there is possibility to tell to when manual flight mode autopilot to no take the rc override command when lag goes up set param (for example RCOVERRIDE_MAX_LATENCY) ? for example 200ms or 500ms. (for example in loiter, UAV hold it's position waiting that lag goes down).

And if lag goes up set another param it trigger failsafe action ?  (for example RCOVERRIDE_FAILSAFE_TRIG) ? for example 600ms - 2 000ms

thank you.

rmackay9, 2020-04-13:
@roque-canales, this doesn't seem correct to me.  If the RC failsafe (aka throttle failsafe) is enabled it should catch a laggy input from the pilot regardless of whether a regular TX/RX is used or a joystick connected to the GCS (aka "RC_OVERRIDES").

I think I'll close this issue but I'll keep monitoring for replies.

closed 2020-04-13 by rmackay9

roque-canales, 2020-04-13:
Thank you for this reply Randy, ok I will test with throttle failsafe activated.
Can you tell me if its possible to set the timeout ? Or what is hardcoded?
Thank you.