ArduPilot/ardupilot #14067: Mavlink telemetry limited to about 1450 bytes per second on a 19200 bauds port

tadam777, 2020-04-12:
## Bug report

I noticed that in recent versions of APM Plane, the mavlink output on a TELEM port is unable to use the full 19200 bauds bandwidth.

How to reproduce it :
1) Connect an FTDI adapter on a telemetry port
2) Set the port to 19200 bauds
3) Connect with Mission Planner, APM Planner or another GCS through this FTDI adapter.
4) Change the telemetry rates, and observe the "bytes per second" in the "Stats" window of Mission Planner. Even with huge telemetry settings such as requesting 100 times per second the Attitude, the bytes per second stay around 1450 bytes per second with the APM Plane 4.0.5

The behaviour in older versions of APM Plane was to be around 1920 bytes per second, which is the limit for 19200 bauds. As an example I've made a video a few years ago with APM Plane 3.3.0 and we can for example see in the video the bytes per second to be around 1700 bytes per second and that wasn't the limit : https://vimeo.com/137746307 at 1:11 in the video.

The behaviour is already present in APM Plane 3.9.8, but in that version the limit is, around 1550 bytes per second.

I tried a lot of telemetry rates combinations, these values seem to be the limiting factor, and depend on the version.

I can reproduce it either in direct (with an FTDI adapter) or with a radio such as Ultimate LRS, the behaviour comes from APM Plane and not from the link. ULRS allows also to visualize the data rate, and confirms it's now limited around 1450 bytes per second.

No packet loss is observed, which seems to confirm that the flight controller isn't trying to send more data that would be lost for some reason (ftdi or radio issue for example).

I'm the developper of Ultimate LRS, an LRS radio system that's often used with Pixhawk, so I'm used to work in detail with the telemetry and its speed.

But there's a secondary issue apart from the speed impact : if the bytes per second stay below the 1450 bps mark (by choosing low telemetry rates), everything is very responsive, especially the artificial horizon moves immediately without delay.
But once the limit of 1450 bps is reached, because we've selected higher telemetry rates, the bps will stay at 1450 bps, but a 1 second delay appears in the movements of the artificial horizon. This is probably related to the mavlink message queue in the flight controller which becomes full (new "attitude" messages are dropped, and "old" attitude messages are sent first.

This is just to explain that it's not only a speed issue for the users, but also creates a delay issue, which makes the artificial horizon move with a 1 second delay, making it difficult to fly just looking at the artificial horizon. It's difficult to understand for the users, because they see a 1450 bps data rate, which is quite below the 1920 bps (19200 bauds) limit they're used to since years.

I didn't find any related issue.

**Version**

I can confirm the issue exists in APM Plane 4.0.5 (latest version for the moment) on Pixhawk and APM Plane 3.9.8 on Pixhawk. It wasn't present in APM Plane 3.3.0 (it was tested years ago on an APM 2.7 board, but not sure if it depends on the board).

I'll try to identify in which version it started to appear.

**Platform**
[  ] All
[  ] AntennaTracker
[  ] Copter
[X] Plane
[  ] Rover
[  ] Submarine

I didn't test on other platforms, but suspect that it may be present as well.

**Airframe type**
_What type of airframe (flying wing, glider, hex, Y6, octa etc)_

**Hardware type**
Issue present with Pixhawk for versions APM Plane 3.9.8 and 4.0.5
Tested years ago on APM 2.7 with version APM Plane 3.3.0 without encoutering this issue.

**Logs**

tadam777, 2020-04-13:
I tried today to disable the other telemetry ports, because maybe the CPU was overloaded by computing mavlink packets for all ports.

This improved the bytes per second which now reaches 1730, but this is still lower than what was possible on previous versions (about 1920 bytes per second).

I made a video showing where the speed is visible and even during parameters transmission it maxes at 1730 (the video wasn't made specifically for this issue, but shows the issue) :

https://vimeo.com/manage/videos

athertop, 2020-04-20:
Keen to see the outcome of this one. Hoping it might be picked up soon.

peterbarker, 2020-04-20:
See https://github.com/ArduPilot/ardupilot/pull/14119

tadam777, 2020-04-21:
I'm not sure the cause is related to a CPU overload, because it happens for small telemetry rates, as long as they go over 1450 bytes per second.

You can see an example here with the same telemetry rates, for APM Plane 3.8.5 and for APM Plane 4.0.5

In 3.8.5 the total telemetry is 1790 bytes per second (and if choosing higher telemetry rates we can go up to 1920 bytes per second). We can also see that the received telemetry matches the requested telemetry rates.

![2020-04-21 09_49_10-Mission Planner 1 3 62 build 1 3 6917 15581 ArduPlane V3 8 5 (7dc5e040)]
(https://user-images.githubusercontent.com/4885885/79842590-5404a380-83b9-11ea-9788-8c2cb15df305.png)

In 4.0.5 (and since 3.9.0) the total telemetry is 1462 bytes per second, and we can see that the received telemetry doesn't match the requested rates. Attitude messages are received only 4 times a second rather than 7 times.

[image]

peterbarker, 2020-04-21:
It isn't CPU overload; it's running out of time to send messages in the allotted timeslice.

Try pushing your `SCHED_LOOP_RATE` up - that adds more timeslices/second, so you should see your telemetry rates improve.

tadam777, 2020-04-21:
Parameter SCHED_LOOP_RATE doesn't solve the issue, on the contrary, the telemetry rate decreases when I increase SCHED_LOOP_RATE above 50.

To test it I set up the telemetry rates in Mission Planner to only request Position at 50 Hz (which is more than what can be sent at 19200 bauds)

With APM Plane 4.0.5 (waiting about one minute after startup)
SCHED_LOOP_RATE=50 --> 1470 bytes per second
SCHED_LOOP_RATE=100 --> 1370 bytes per second
SCHED_LOOP_RATE=200 --> 1400 bytes per second
SCHED_LOOP_RATE=10 (to try) --> 1300 bytes per second

In 3.8.5 I don't know if this parameter is taken into account, but I tested with :
SCHED_LOOP_RATE=10 --> 1920 bytes per second (full 19200 bandwidth is used)
SCHED_LOOP_RATE=100 --> 1920 bytes per second (full 19200 bandwidth is used)

peterbarker, 2020-04-21:
On Tue, 21 Apr 2020, tadam777 wrote:

> Parameter SCHED_LOOP_RATE doesn't solve the issue, on the contrary, the
> telemetry rate decreases when I increase SCHED_LOOP_RATE above 50.

OK, thanks for testing.

So my next best guess is that this is the stream-slowdown code being 
somewhat over-zealous.

Are you using hardware flowcontrol on these links?

Check the "MAV" dataflash log message - is "MAV[x].ss" non-zero on the 
relevant links (where "x" is the link number; 0 for USB, 1 for serial1, 
generally)

tadam777, 2020-04-23:
It's very possible that it's the stream-slowdown code, as there's no packet loss. I'm not using hardware or software flow control.

You'll find attached the log, running 4.0.5 with 19200 bauds telemetry connected on telem1, while the total bps is limited to 1450 bps. The same setup give more than 1700 bps with these SR settings, and can go up to 1920 bps in 3.8.5 with higher SR settings.

[2020-04-23 23-23-01.log](https://github.com/ArduPilot/ardupilot/files/4525418/2020-04-23.23-23-01.log)

peterbarker, 2020-04-23:
@tadam777 Could you get me a log with master running on this - or the "latest" build, please?

tadam777, 2020-04-24:
I see that you added some specific mavlink statistics to the SD logging recently, thanks for that.

You'll find here this morning's master running at telemetry rates 7 / 5 / 1 / 1 in Mission Planner.

[00000003.zip](https://github.com/ArduPilot/ardupilot/files/4528055/00000003.zip)

And the same running at telemetry rates 1 / 1 / 1 / 1 which gives a very low telemetry below 1000 bps.

[00000003.zip](https://github.com/ArduPilot/ardupilot/files/4528091/00000003.zip)

tadam777, 2020-04-24:
I did additional tests by showing the txspace() value in the SD logs. Then requesting the attitude at 10 / 5 / 1 / 1 / 1, which is more than what's possible on 19200 bauds. However it gives only 1280 bytes per second (tested on the master).

I also notice that before there's the GPS fix, the throughput is around 1670 bytes per second, but as soon as the GPS fix is there it goes down to 1280.

What happens is that the txspace() shows that the tx buffer is constantly full (less than 100 bytes available, around 50 bytes left on average). I also checked the txspace every 100 ms to be sure it's not sometimes free again.

So the question is : why, with the tx buffer constantly full, the resulting output is only around 1280 bytes per second ? And why is there a difference when we have a GPS fix ? could there be an interaction as the GPS is also connected on an UART ? Is it a threading issue ?

tadam777, 2020-04-24:
Still another test, I modified the UART_test.cpp example to send a lot of data on TELEM1, and it goes up to 1900 bytes per second without problem, so the UART library works fine and doesn't limit the rate.

peterbarker, 2020-04-26:
> It's very possible that it's the stream-slowdown code, as there's no packet loss. I'm not using hardware or software flow control.

Having looked at your logs, it's not stream slowdown.  Whatever manner of telemetry you are using, it doesn't provide feedback (via `RADIO_STATUS`) on any external buffer (which we don't want to overflow) like the SiK radio firmware does.  That's what stream-slowdown is based on.

peterbarker, 2020-04-26:
> I did additional tests by showing the txspace() value in the SD logs. Then requesting the attitude at 

That, and a few other statistics, are slated for a `UART` message.
 
> I also notice that before there's the GPS fix, the throughput is around 1670 bytes per second, but as soon as the GPS fix is there it goes down to 1280.

OK, that's bad.  Your logs don't *seem* to show extreme CPU load, which would be the obvious thing to look at if something changes when we go to full-fusion.

> What happens is that the txspace() shows that the tx buffer is constantly full (less than 100 bytes available, around 50 bytes left on average). I also checked the txspace every 100 ms to be sure it's not sometimes free again.

And *that* is truly weird, especially when combined with your observation about the example working just fine.  Only thing I've thought of there is the buffer size - but I think both should be using the same buffer size.  Given the atypically low baudrate, however, I would suggest one test you could do is to increase the size of the uart buffer for that serial port and see what happens.

tadam777, 2020-04-26:
> Having looked at your logs, it's not stream slowdown. Whatever manner of telemetry you are using, it doesn't provide feedback (via `RADIO_STATUS`) on any external buffer (which we don't want to [...]

OK, you're right there's no RADIO_STATUS messages because ULRS can handle the full 19200 bauds and doesn't need to slow it down. I thought you were referring to the mechanism of the deferred messages.
Actually the reason it runs at 19200 bauds is because it's a good rate for mavlink, it's for example possible to have 5 Hz attitude and position messages, and the other messages at 1 Hz. And having the flight controller sending at 19200 bauds avoids the need for hardware or software flow control.

tadam777, 2020-04-26:
> That, and a few other statistics, are slated for a `UART` message.
>
>
> OK, that's bad. Your logs don't _seem_ to show extreme CPU load, which would be the obvious thing to look at if something changes [...]

It's a good idea, I'll try that. For the moment it has a 1024 bytes buffer.

tadam777, 2020-04-28:
I've made the same test with different versions, this time just connected via USB so anyone can do the same test. No GPS is connected (because more messages would be sent when the GPS fix happens). It shows somehow that the older versions are much faster with telemetry.

I'm requesting telemetry at the maximum rates allowed by mission planner, and show the total bytes per second, and packets per second for versions 4.0.5, 3.8.5 and 3.8.0

Version 3.8.0 is three times faster than the latest (4.0.5) version, and version 3.8.5 is twice faster than the latest version.

I'm not sure if this is related to this issue, but obviously something changed between the versions. I don't know what limits the rate in the end.

The firmware version can be seen at the top of each screenshot :

[image]
[image]
[image]

david9981, 2021-11-29:
I came across this thread when trying to investigate why I am getting huge packet loss of up to 70% as shown in QGC, and link quality gcs as low as 30 in Mission Planner.  I am using SiyiFM30 rx and the default baudrate is 57600.  Looking at Mavlink Inspector, the data stream is only 1966bps, and after GPS fix, dropping to 1553bps.

I get the same result in both Mission Planner and QControl, so I don’t think it is ground station software related. I am only running Arduplane so I can’t compare the results to iNav/BF. But I’ve tried everything: swapping telem1 with telem2, different UARTs, different serial baudrate, adjusting the telemetry stream rates in both MP and QGC (all to 1Hz), adjusting SRx_ parameters (slight improvement by setting everything to 1), and [adjusting](url) the SCHED_LOOP_RATE. No luck. Any ideas please?
[image]

closed 2025-07-19 by tadam777