ArduPilot/ardupilot #14069: problema multi MaxbotixI2C in arducopter 4.0.3

robynat, 2020-04-12:
Impostare piu MaxbotixI2C su arducopter 4.0.3 non funziona poichè anche se si cambiano gli indirizzi di ogni singolo modulo sonar e vengono inseriti nelle impostazioni  di arducopter 4.0.3 non funzionano poichè viene letto solo l'indirizzo di default (224) e non viene tenuto conto dei nuovi indirizzi impostati....mi chiedo se è possibile ottenere una correzione del firmware per gestire diversi di questi sonar contemporaneamente.....grazie
If there was a previous discussion in the forum, link to it

## Bug report

**Issue details**

_Please describe the problem_

**Version**
_What version was the issue encountered with_

**Platform**
[  ] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
_What type of airframe (flying wing, glider, hex, Y6, octa etc)_

**Hardware type**
_What autopilot hardware was used? (Pixhawk, Cube, Pixracer, Navio2, etc)_

**Logs**
_Please provide a link to any relevant logs that show the issue_

rmackay9, 2020-04-13:
I've updated the subject line and contents a bit to translate to english but also specify what we need to do to support this.  It's not a big change actually.

robynat, 2020-04-13:
ok....di certo per voi non è un grosso problema mentre per me è quasi invalicabile non conoscendo questo genere di programmazione .....resto in attesa di un vostro aggiornamento al firmware.......se invece avete qualche dubbio nel campo elettronico (e non credo visto lo sviluppo  continuo e costante del firmware) io potrei aiutarvi come elettronico specializzato in microelettronica in avionica.....grazie

robynat, 2020-04-13:
Comunque ho costruito 3 droni con tre diversi fly controller ed oltre questo problema non ho riscontrato nessun'altra anomalia nel firmware 4.0.3 ...almeno per come li uso io...ancora grazie

hendjoshsr71, 2021-09-29:
Closing. This was fixed here https://github.com/ArduPilot/ardupilot/pull/14636

closed 2021-09-29 by hendjoshsr71