ArduPilot/ardupilot #14075: EKF3: origin cannot be set unless GPS is enabled

rmackay9, 2020-04-13:
When testing the Intel RealSense T265 using Thien's scripts that can now output position-delta messages, I found that the EKF3 was refusing commands to set the origin.  The message, "EKF3 refusing set origin" was displayed.

I had to momentarily re-enable the GPS (by setting GPS_TYPE = 2), the origin was then set and then I could disable the GPS again and rely only on position-delta data.

The EKF3 should be able to have it's origin set without a GPS enabled.  I'm pretty sure that it could do this at some point, so I think this is technically a regression.

anbello, 2020-04-13:
With EK3_GPS_TYPE = 3 you should be able to set origin even if GPS_TYPE = 0

rmackay9, 2020-04-14:
@anbello, thanks very much for the correction! I've retested with EK3_GPS_TYPE = 3 I was able to set the EKF origin.

closed 2020-04-14 by rmackay9

romaster93, 2021-02-09:
i cant find a EK3_GPS_TYPE.....what should i do? 

now im using copter 4.1.0-dev

rmackay9, 2021-02-10:
@romaster93,

It's best to [create a new topic in discuss](https://discuss.ardupilot.org/c/arducopter/copter-40/155) or join us in the "viision-projects" channel in discord ([see link on this page](https://ardupilot.org/copter/docs/common-contact-us.html))

standmit, 2021-12-14:
Origin cannot be set until EK3_SRCn_POSXY = 3
https://github.com/ArduPilot/ardupilot/blob/03bf772aaf8deb2edc6222578339c8999454795b/libraries/AP_NavEKF3/AP_NavEKF3.cpp#L1362-L1369

rmackay9, 2021-12-14:
@standmit,

Just to clarify, the EKF origin can**not** be set if EK3_SRCn_POSXY = 3.

hi6otb, 2022-09-22:
> hi, i am trying autonomous flight indoor with Pozyx anchors. I have the same problem, the EKF3 refuses the origin. I have GPS_TYPE = 0, and in EKF3_SRCn_POSXY = 4. Is this correct? The drone refuses to start any mission because it sens "position required"

Were you able to resolve it?