ArduPilot/ardupilot #14081: Rover: DPTH log message should include the rangefinder instance

rmackay9, 2020-04-14:
If a user sets up two (or more) downward facing echosounders, Rover will log two DPTH messages (one for each) but it won't be possible to determine which is which.  We should add the rangefinder's instance to the DPTH message to resolve this.

shiv-tyagi, 2021-09-22:
Hey! I would like to work on this issue. I poked around and found the method which we are using to log depth. I can't understand how we are coding units while passing them to that method and what do `mults`, `fmt` mean. Also, which rangefinder property should I log to differentiate between multiple rangefinder instances. Please point me in right direction. Thanks in advance.

IamPete1, 2021-10-03:
fixed by https://github.com/ArduPilot/ardupilot/pull/18751 thanks @shiv-tyagi !

closed 2021-10-03 by IamPete1