ArduPilot/ardupilot #14091: Renaming internal vehicle names pain points

peterbarker, 2020-04-15:
## Tracking Document

This document simple seeks to record the odd and/or unexpected things that break when you do things like rename a vehicle from e.g. APMrover2 -> Rover

 - autotest.ardupilot.org needs a redirect added for the Parameters file; look for the APMrover2 redirects
 - the github tags still use the full old name.  So, for example `APMrover2-stable`.  Currently there are patches in build_binaries which map back from Rover -> APMrover2 for the purposes of checking out firmware based on tags

brunoolivieri, 2021-04-11:
Was there no adjustment in the parameter version script?

peterbarker, 2021-04-11:
@brunoolivieri I believe the versioning script was written after the rewrite - but thanks for mentioning it here so we don't forget it for the other vehicles!

peterbarker, 2021-05-10:
MissionPlanner is sensitive to file name changes on our SITL binaries.

This broke things for `apmrover2.exe` -> `ardurover.exe` - and patch was required to do an additional copy step.

However, other vehicle renames are unlikely to have this problem as their binary names already look correct.