ArduPilot/ardupilot #14105: Copter: Failsafe not working correctly with inverted input on serial port.

JacksonUAS, 2020-04-16:
## Bug report

**Issue details**

Putting inverted SBUS onto a serial port as per multiple Rx install and enabling FS_THR_ENABLE results in looping failsafe alarm. FS_THR_ENABLE disabled results in no failsafe.

**Version**
AC4.0.3

**Platform**
[  ] All
[  ] AntennaTracker
[ X ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
X8

**Hardware type**
Cube Black
Herelink (inverted SBUS onto Serial 5/CONS)

**Logs**
https://discuss.ardupilot.org/t/dual-rx-input-insanity-herelink-and-rfd900x-sbus-passthrough/54620/8

JacksonUAS, 2020-05-01:
Transposed SBUS inputs, can confirm there is a bug with the handling of failsafe condition on inverted serial receiver.

JacksonUAS, 2020-05-05:
Issue exists on cube orange without external SBUS inversion as well.

Pedals2Paddles, 2020-05-05:
Based on the log provided, the problem appears to be AP processing the RC input from the serial port.  The log shows the channels all bouncing between good and no signal.  This of course is making the failsafe kick on and off. It is NOT a failsafe problem.  The failsafe is just a symptom of the RC handling problem.  The user's log is attached.

He has already tested using multiple different inverters and difference receivers. Everything looks fine on a scope.  But AP shows the input bouncing all over the place.  Given his multiple methods of testing to rule out a hardware issue, I'm going to tag this as a bug.  Feature originally added here last year https://github.com/ArduPilot/ardupilot/pull/12152.

Log attached from @JacksonUAS 
[serialRCinput.zip](https://github.com/ArduPilot/ardupilot/files/4583697/serialRCinput.zip)

Pedals2Paddles, 2020-05-05:
@JacksonUAS I changed the title since this is a core RC protocol library problem that impacts all vehicles and builds, not just a copter failsafe issue.

JacksonUAS, 2020-06-05:
Any updates on this? Is there any timeline I can convey to operators about when this will be investigated.

sachman, 2020-08-10:
Any update on this? This seems like a very serious bug,

sachman, 2020-08-14:
Could it be a problem of SBUS data capture itself (due to CPU overload)? Or of processing?

sachman, 2020-08-14:
I have increased the RCIN logging rate to 400 Hz (in code). Below is a screenshot of what is typically happening to RC signals on my side. Might be helpful in debugging.  The frequency of these glitches are quite high.
[image]

[image]

Also, I have seen glitches, although really rarely, on Copter 3.5.5.

sachman, 2020-08-16:
The problems seems to be of data capture:
With enabling DMA for the RX of the uart port capturing SBUS, the glitches reduces by around 20 times (and hence so does desired attitude and other relevant aspects.)

Attaching here logs for the two setups from my test bench: RCin logging at 400 Hz,  Attitude Logging at 400 Hz.

https://drive.google.com/file/d/1RgvqAzRRvqQUCrWIuBeJ17MWSuLqQQvb/view?usp=sharing

I don't see these kind of glitches on Copter 3.5.5 as significant as this.

Has the CPU processing requirements increased from 3.5.5 to 4.0.3??

sachman, 2020-08-17:
With DMA enabled, I tried increasing the DMA stream for the relevant USART to highest (from 0 to 3). However, that also did not completely remove the RCin glitches. I am super confused right now. Can somebody point me any further?

sachman, 2020-08-26:
Hi,
The problems seems to be of capture and reading pipelines - the capture (when DMA is not enabled) is poll based in UART thread which has priority of 60 (low priority), and rcin reading in RC thread (which has lower priority than the main ap thread) are too slow. 
If you move the reading part of RC from RC thread to TImer thread, and if you enable DMA on RX or capture byte in interrupt callback for the uart, the problem goes away. No glitch at all!!!!!

JacksonUAS, 2020-09-19:
Any updates on this one? I have the aircraft back here to try sort out for customer.

andyp1per, 2020-09-19:
Use a DMA enabled serial port seems like the safest option. Changing the thread priority of the RCIN or UARTs is a more intrusive change. I suspect this is true for a lot of RC protocols - we should probably add DMA to the decoding report so that people know.

peterbarker, 2023-12-13:
@JacksonUAS this one seems to have dropped off the radar for 3 years.  Is this actually still a problem?!

peterbarker, 2024-07-31:
No answer in half a year.

We've completely rewritten that handling several times since 2020.  I don't think its realistic we'll trace this problem now as it has almost certainly been replaced with better bugs.

Closing this now.

closed 2024-07-31 by peterbarker