ArduPilot/ardupilot #14111: Support FrSky OSD / ANALOG PIXEL OSD

MFlghts, 2020-04-16:
## Feature request

**Platform**
Plane
Copter

A very nice pixel osd was released. It's better than old MAX osd in many ways. It's already supported by Inav and all implementation can be ported I guess.

It will give great new possibilities, and finally solve the problem of using old ugly osd that many people were not happy about.

https://www.frsky-rc.com/product/osd/
http://www.mateksys.com/?portfolio=pxosd

SergeyBokhantsev, 2020-04-19:
@MFlghts do you know an examples in youtube of how it looks on INav? I found only one short demo and I was not impressed of the look and performance (frame rate was low). Also an artificial horizon was the only graphical element there.

I've played with this OSD and it has a potential indeed but there are some problems I found:
 1. Firmware has some bugs related to drawing functionality, FrSky didn't provide any advice on those, nor if they have any plans for a FW upgrades.
 2. API is available here https://github.com/FrSkyRC/PixelOSD/blob/master/doc/OSD.md but again, FrSky doesn't provide any extra clarification of the functionality that isn't covered completely in this doc.
 3. The UART port at 115K is a bottleneck. If you want to implement a nice looking, responsive, rich graphical picture - then you need to make a lot of optimization in your drawing code. Otherwise you just have no throughput enough, and picture is laggy. For graphical elements a refresh rate of 20Hz is a minimum, so throughput utilization grows very fast with any extra OSD graphical element.
 4. Thin lines sometime looks ugly at some angle (around 45 degrees). Same problem for a char fonts that have skinny symbols, (like https://github.com/ArduPilot/ardupilot/blob/master/libraries/AP_OSD/fonts/bfstyle.png) - some symbols almost not readable. Though not a problem for a normal fonts.
 5. Image artifacts (shadows). May be not a FrSky issue but a nature of analog signal. Horizontal lines has a white "shadows" by the right side.  

Just my little experience. 

Below is the my experiment, this utilizes about 10% of UART throughput
https://www.youtube.com/watch?v=HICAp-99wCQ

MFlghts, 2020-04-19:
Thank you for deep investigation. This is very sad that FRSKy keeps producing unfinished, uncompleted and undebugged products. If they will not provide proper support it might be considered that this OSD is the dead end, and can be used only for some low-end projects like Inav.

fiam, 2020-04-21:
@SergeyBokhantsev Just to clarify: I'm not an FrSky employee, but I did the OSD project for them. There were more features I wanted to include and some things that I wanted to polish more, but due to time constraints I wasn't able to include them in the 1.0 release. Note, however, that there are updates planned, and the app for upgrading the OSD is even open source. You can check it out here https://github.com/FrSkyRC/FrSkyOSDApp. I guess you could take a guess given the commit dates in that repo, but I can't confirm exactly when a new firmware version will come out.

Regarding the API documentation, there's a command for changing the baudrate at https://github.com/FrSkyRC/PixelOSD/blob/master/doc/OSD.md#osd_cmd_set_data_rate--122. I've tested it up to 921600 and it works fine, but I didn't set any explicit limits and higher data rates might work too.

If you have any questions regarding the API, suggestions about new functionality or bugs that you'd like to report, please feel free to drop me an email to alberto@garciahierro.com

SergeyBokhantsev, 2020-04-21:
@fiam great, thanks. 
The baudrate increasing was in my mind, but I was afraid to brick my board. Didn't get an answer from FrSky on this. Good to hear it can work at that high speed. This shall eliminate the bottleneck problem I faced.

fiam, 2020-04-21:
@SergeyBokhantsev I'll clarify that in the documentation, thanks for pointing it out. The change to the data rate is not persistent, when you reboot the OSD it will always start at 115200. Note that if you're writing a driver for an FC, you need to take into account the case when just the FC is rebooted without restarting the OSD.

Linjieqiang, 2020-08-14:
@fiam Hello fiam. Long time no see. I want to know that the Frsky OSD is now supporting the mavlink2 protocol for Ardupilot FC?

sixtyfive, 2023-08-26:
@flam sadly the documentation (well, the whole fw repo) has been taken offline by them. any chance a copy might still, hmm, be out there, somewhere?