ArduPilot/ardupilot #14118: IIC doesn

liang-tang, 2020-04-17:
**IF YOU DON'T REMOVE THESE FOUR LINES, THEN YOUR REPORT WILL BE CLOSED AUTOMATICALLY**
Questions and user problems should be directed at the forum (http://discuss.ardupilot.org)
_**Please be very sure you have found a bug when opening this issue**_
If there was a previous discussion in the forum, link to it

## Bug report

**Issue details**

_Please describe the problem_

**Version**
_What version was the issue encountered with_

**Platform**
[  ] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
_What type of airframe (flying wing, glider, hex, Y6, octa etc)_

**Hardware type**
_What autopilot hardware was used? (Pixhawk, Cube, Pixracer, Navio2, etc)_

**Logs**
_Please provide a link to any relevant logs that show the issue_

rmackay9, 2020-04-17:
@liang-tang, can you clarify a bit more what the issue is?  Are you saying that I2C never works?  Is it failing with just some sensors?  Is it failing to do some operation but it is difficult to notice?

liang-tang, 2020-04-17:
@rmackay9 updated comment

liang-tang, 2020-04-17:
> @liang-tang, can you clarify a bit more what the issue is? Are you saying that I2C never works? Is it failing with just some sensors? Is it failing to do some operation but it is difficult to notice?
Frist, I added debug message like below,
[image]

Second, reset all params, just set RNGFND1_TYPE to 16, and reboot.
Thrid, connect fcu to serial tool, and got the log,  that means iic return fase when check VL53L1x's ID

> Successfully mounted SDCard (slowdown=0)
> 
> 
> Init ArduCopter V4.0.3 (ffd08628)
> 
> Free RAM: 154328
> FW Ver: 120
> ----------------------------------------
> 
> 
> load_all took 1532us
> 0 0 0 IOMCU: CRC ok
> Detected Pixhawk
> IOMCU startup
> _transfer 310
> _transfer 310
> _transfer 321
> _transfer 321
> _transfer 321
> _transfer 321
> _transfer 321
> MS5611 found on bus 1 address 0x03
> AP_Logger_File: buffer size=16384
> _transfer 310
> _transfer 310
> _transfer 310
> _transfer 310
> ?_transfer 310
> _transfer 310
> disabling flow control on serial 2
> disabling flow control on serial 3
> _transfer 310
> ??_transfer 310
> _transfer 310
> _transfer 310
> _transfer 310
> ?_transfer 321
> _transfer 321
> _transfer 321
> _transfer 321
> _transfer 321
> _transfer 321
> _transfer 321
> _transfer 321
> _transfer 321
> _transfer 321
> _transfer 321
> AK09916: Reset Failed
> _transfer 321
> AK09916: Reset Failed
> _transfer 321
> _transfer 321
> _transfer 321
> _transfer 321
> _transfer 321
> _transfer 321
> _transfer 321
> _transfer 321
> _transfer 321
> _transfer 321
> _transfer 321
> ?_transfer 310
> _transfer 310
> _transfer 310
> _transfer 310
> ??_transfer 310
> _transfer 310
> _transfer 310
> _transfer 321
> _transfer 321
> 000------
> _transfer 321
> _transfer 321
> 000------
> _transfer 310
> ?Init Gyro_transfer 310
> *_transfer 310
> _transfer 310
> ?_transfer 310
> _transfer 310
> _transfer 310
> *_transfer 310
> _transfer 310
> _transfer 310
> *_transfer 310
> _transfer 310
> ?	*_transfer 310
> _transfer 310
> _transfer 310
> ?
> 
> 
> Ready to FLY _transfer 310
> _transfer 310
> _transfer 310
> _transfer 310
> ?_transfer 310
> _transfer 310
> _transfer 310
> _transfer 310
> ?_transfer 310
> _transfer 310
> _transfer 310
> _transfer 310
> ?
> _transfer 310
> _transfer 310
> _transfer 310
> _transfer 310
> ??o_transfer 310
> _transfer 310
> _transfer 310
> _transfer 310
> ?_transfer 310
> _transfer 310
> _transfer 310
> _transfer 310
> ?_transfer 310
> _transfer 310
> _transfer 310
> _transfer 310
> ?_transfer 310
> _transfer 310
> _transfer 310
> _transfer 310
> ?_transfer 310
> _transfer 310
> _transfer 310
> _transfer 310
> ?_transfer 310
> _transfer 310
> _transfer 310
> _transfer 310
> ?_transfer 310
> _transfer 310
> _transfer 310
> _transfer 310
> ?_transfer 310
> _transfer 310
> _transfer 310
> _transfer 310
> ?_transfer 310
> _transfer 310
> _transfer 310
> _transfer 310
> ?_transfer 310
> _transfer 310
> _transfer 310
> _transfer 310
> ?_transfer 310
> ?o_transfer 310
> _transfer 310
> _transfer 310
> ?_transfer 310
> _transfer 310
> _transfer 310
> _transfer 310
> ?_transfer 310
> _transfer 310
> _transfer 310
> _transfer 310
>

liang-tang, 2020-04-17:
works well now, because I forget set IIC addr....:: "><"

closed 2020-04-17 by liang-tang