ArduPilot/ardupilot #14139: Enable hal.console->printf

HefnySco, 2020-04-20:
https://github.com/ArduPilot/ardupilot/blob/bd688ef4121dc511004eb8438d24ac1e3989afb2/libraries/GCS_MAVLink/GCS_Common.cpp#L928

by commenting:
_#if CONFIG_HAL_BOARD == HAL_BOARD_SITL
        AP_HAL::panic("next_deferred_bucket_message_to_send called on empty bucket");
#endif_ 
I was able to call **hal.console.printf** in modules such as AP_GPS.cpp while running SITL and every thing worked well.

I wonder if we should remove this condition or refine it ?
kindly advise.

IamPete1, 2021-08-16:
I wonder if this has been resolved by https://github.com/ArduPilot/ardupilot/pull/14743 ?

HefnySco, 2021-08-16:
It is working now. Not sure why because even if I remove #14743  updates it still works.

closed 2021-08-16 by HefnySco