ArduPilot/ardupilot #14168: Flash Storage example failing

peterbarker, 2020-04-23:
## Bug report

**Issue details**

Running
```
./waf configure --debug --board=linux && ./waf build --target=examples/FlashTest && ./build/linux/examples/FlashTest
```

yields
```
WARNING: not running as root. Will not use realtime scheduling
FATAL: invalid write16 at 0:4 0x00b1 0x00b0
```

**Version**
master

**Platform**
[  ] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine
[X] Example

**Airframe type**
n/a

**Hardware type**
SITL

**Logs**

n/a

tadam777, 2020-04-26:
I confirm the same message "FATAL: invalid write16 at 0:4 0x00B1 0x00B0" on a Pixhawk.

peterbarker, 2020-06-29:
Thanks @tadam777 - good to have confirmation.

peterbarker, 2025-11-10:
Still failing in Nov 2025 :-)

LupusTheCanine, 2025-11-12:
> This got me confused — what should we do? According to me, the execution is just fine; it’s just a test case that is meant to fail!

We could force seed that doesn't trigger multiwrite failure 😅
Or even better, have two tests, one with current seed and the other with safe one 😁