ArduPilot/ardupilot #14173: ArduCopter doesn't initialize unless barometer is present (even if not used)

tenten8401, 2020-04-23:
##  Bug report

**Issue details**
When plugging in my Matek F405 CTR running 4.0.3 on MatekF405 target, the board does not finish initializing unless a barometer is present. I de-soldered the barometer on this board because I conformal coated it by mistake and didn't want it producing weird readings. While I agree that having the barometer missing isn't a common thing there should at least be a way to disable it to prevent it from stalling the boot process and preventing me from doing anything..
EK2_ALT_SOURCE = GPS

**Version**
ArduPilot 4.0.3

**Platform**
[  ] All
[  ] AntennaTracker
[X] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
Quad

**Hardware type**
Matek F405 CTR

**Logs**
[image]

tenten8401, 2020-05-07:
I think this may be considered a bug since the MatekF405 target has this line in it and it's not working:
https://github.com/ArduPilot/ardupilot/blob/master/libraries/AP_HAL_ChibiOS/hwdef/MatekF405/hwdef.dat#L136
```
define HAL_BARO_ALLOW_INIT_NO_BARO
```

tenten8401, 2020-05-08:
This will be fixed by #14317

closed 2020-05-08 by tenten8401

albertocopter, 2020-05-15:
> This will be fixed by #14317

I have had the same problem since I mounted a compass (i2c conflict?). To solve you need to connect the lipo first and then the USB. The problem, however, remains in the reading of the altitude by the barometer because it marks - 6/7 meters also on the ground. This makes flight scheduled on waypoints or RTL inaccurate. Can it be solved?

tenten8401, 2020-05-16:
@albertocopter This issue was specifically for when there is no barometer onboard, not barometer isn't detected -- may be worth opening another issue over