ArduPilot/ardupilot #14175: RC6 tuning for WP_Speed no longer working with AC4.0.x

hsteinhaus, 2020-04-23:
## Bug report
RC6 tuning is no longer working in AC4.0x. To reproduce, launch SITL and do the folloing commands: in the mavproxy console:

```
param set TUNE 10
param set TUNE_MIN 100
param set TUNE_MAX 1000
rc 6 1900
param fetch
param show WPNAV_SPEED
```

The result is always 500, regardless of the value set by "rc 6". Expected value should be around 990.

I have flown the same version in the real world today - same effect, no response to WP speed tuning. This is particularly bad, as this function is a great help to protect the vehicle from overload when fighting against the wind in an lengthy AUTO mission. 

**Version**
4.04-dev ( eb85d98b4d60b60e8acfd71600773a2d254deb1f )

**Platform**
[  ] All
[  ] AntennaTracker
[X] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
X4

**Hardware type**
PixRacer

**Logs**
Not needed - just SITL

JacksonUAS, 2020-04-23:
The parameter name changed, sorry not in front of computer to find the correction, but stumbled on this a few weeks ago.

hsteinhaus, 2020-04-24:
The parameter-related part of the tuning seems to work well. The values are captured from the radio correctly and are applied to the WP_Nav speed. I have successfully traced the tuning value until here: https://github.com/ArduPilot/ardupilot/blob/864717310de95e11b2f8fe4fb29485bf1a86acfe/libraries/AC_WPNav/AC_WPNav.cpp#L158

My first guess is, that the following change might be the one that breaks the WP_Speed tuning:
https://github.com/ArduPilot/ardupilot/commit/593ff6818d0d2540c139d489b9c52c5fa97ab07b

I will do more research tonight and will report back.

rishabsingh3003, 2020-04-24:
The tuning works just fine (At least on SITL) .. Here is a log of GPS speed + Channel 6 input... You can see the increase/decrease with the ch6 input.
[image]

For the changes to reflect in `param show WPNAV_SPEED` you'll have to store the parameter in EEPROM (Which is a slow operation) every time the tuning function from copter is called. And the tuning function is called at 3.3hz , so I don't think there is any good coming from adding this feature.

I hope this solves your problem

hsteinhaus, 2020-04-27:
> I hope this solves your problem
Very interesting - obviously there is a major undocumented change in behaviour compared to ArduCopter 3.6 (which indeed saves tuned values - just confirmed that with SITL). This changed fooled me into believing the tuning did not work.

But if the tuned value does not reflect in the parameter list, how can I find out what the currently used value is, e.g. when tuning PIDs?

IamPete1, 2021-08-08:
you should be able to seen the changes if you re-fetch the param list. IE:
```
param fetch WPNAV_SPEED
param show WPNAV_SPEED
```

closed 2021-08-08 by IamPete1