ArduPilot/ardupilot #14192: AP_Scripting: set_servo_pwm() does not work for elevator

TunaLobster, 2020-04-25:
## Bug report

**Issue details**

Lua scripting error when using `servo.set_output_pwm` to send PWM to function 19 (elevator). I am attempting to create a way to do consistent inputs between different flight dynamics models using the same control inputs. Lua seemed to be a quick and easy way for me to do it without creating a new mode (I'm not that great with C++ yet). The error does not occur with the branch used for #14120. However, the elevator in SITL does not change when looking at the dataflash logs for RCOU.C2 or AETR.Elev. Is this intended behavior?

Line of code that created the error in 4.0.5 and master:
`servo.set_output_pwm(19, 1900)`

Error produced:
`Lua: ./scripts/hello_world.lua:8: bad argument #2`

Line of code that does not provide expect elevator movement on #14120 without error:
`SRV_Channels:set_output_pwm(19, 1900)`

**Version**
Plane 4.0.5, master, and tridge/pr-pr-scripting-servo

**Platform**
[  ] All
[  ] AntennaTracker
[  ] Copter
[ X ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
Fixed wing

**Hardware type**
SITL?

**Logs**
I can provide a log or the script if need be. Not sure how helpful they will be.

rmackay9, 2020-04-25:
I think this is by design actually.  We also don't allow do-set-servo commands to directly affect control surfaces.

WickedShell, 2020-04-25:
#14120 makes a change to allow control of any servo channel (as well as unifying the syntax to be `servo:set_output_pwm()`

kd0aij, 2020-04-25:
It doesn't look like this would be useful for channels like k_elevator which are being written by the set_servos task...

TunaLobster, 2020-04-25:
I will close this since I am attacking the problem from the wrong angle. I noticed the SYSID mode in Copter and thought I could replicate something similar in Lua. I was doing the elevator movement in manual mode with hopes of keeping the controller out of the stability.

I checked with #14120 to make sure the C++ side wasn't possibly going to change in the future.

Could a feature request be to have the ability to set control surface servos in manual mode?

closed 2020-04-25 by TunaLobster

kd0aij, 2020-04-25:
With current functionality, you could "steal" the control surfaces from the attitude controller by messing with the SERVOn_FUNCTION parameters e.g. changing 2 to 94 
This is probably no more scary than being able to arm/disarm or change flight modes from a script :)

TunaLobster, 2020-04-25:
If I did that I would be leaving my pilot in command without a control surface. Changing modes is easier to fix on the RC transmitter with a mode switch than changing SERVOx_FUNCTION from a parameter page. It would work that way, but to me that feels far scarier if my scripts error out.

kd0aij, 2020-04-29:
@TunaLobster Could you elaborate on what you mean by a "way to do consistent inputs between different flight dynamics models using the same control inputs".
What are the flight dynamics models, and what is the end goal that you have in mind?

TunaLobster, 2020-04-29:
Measure the ability of a flight sim to replicate flight dynamics for a small aircraft. There are plenty of simulators available to use and each one has it's own quirks. I was using ArduPlane to input a control surface deflection that would be equal between the virtual model of an aircraft and a real model. That would allow me to do parameter estimation for each model.

Rather than having a pilot sit in from a computer and try to do the same input many times I was going to use Lua to do it. Hopefully with ArudPlane 4.1 I will be able to do it without jumping into making a new mode in C++.

After your last comment I wrote a Mission Planner script to be ready to send the correct servo function values if the Lua script failed for whatever reason. Small steps towards the end goal.

kd0aij, 2020-04-30:
Wouldn't running autotune in SITL and on the real vehicle let you do that by comparing the resulting logs?
I'm not clear on how/whether one could have a "recovery" LUA script to deal with a scripting crash/hang.  Your approach of having a GCS ready sounds like a useful (if scary) technique.

TunaLobster, 2020-04-30:
Autotune has it's own intricacies that I am trying to keep out of the experiment and doesn't have consistency of control inputs. System identification is great, but on some systems flying without any initial guess of the dynamics is not feasible. How good is the guess that someone might get from SITL simulation though?

If I were to change the servo function from a control surface to a script output using a Lua script and my Lua script hits an error/get dumped due to memory limitations, there is now an aircraft in the air that is missing a control surface. If I set it up such that my GCS Python script triggers when my Lua script fails to report that it has successfully completed, I can then send the correct servo function parameters to the vehicle and jump to RTL mode much quicker than I could clicking around with a mouse or typing on a GCS.

kd0aij, 2020-04-30:
Cool, I hadn't thought of having the GCS react automatically. Just need a solid telemetry link then.