ArduPilot/ardupilot #14204: Copter: terrain following improvements

rmackay9, 2020-04-27:
This is a top-level issue to capture a list of all the ways that terrain following could be improved in autonomous (Auto, Guided) and non-autonomous modes (like Loiter):

- Copter: Terrain data missing for finland. https://github.com/ArduPilot/ardupilot/issues/12923
- Copter: Terrain following/surface tracking in AUTO mode uses unfiltered rangefinder reading. https://github.com/ArduPilot/ardupilot/issues/12911
- Copter: allow terrain following in Auto to mix range finder and terrain database altitudes. https://github.com/ArduPilot/ardupilot/issues/10257
- Copter: using proximity sensor downward oriented for terrain following. https://github.com/ArduPilot/ardupilot/issues/8828

AlexDutilh, 2020-07-02:
Hello,

We are interested in surface tracking in AUTO mode uses rangefinder, in version 3.6 rangefinder filters are available.

Are there any advancements for re-implemented in the following versions (4.0)

thank you

chris89116, 2020-09-07:
Would it be possible initially to only activate the RNGfnd_Gain in Auto mode in order to have this option quickly?
Thank you

MikeCharikov, 2020-12-01:
@rmackay9 

BTW it would be great to have RANGEFINDER_WPNAV_FILT_HZ not hardcoded but to have it as ardupilot parameter. It is not critical but may be good feature for users.

Here is the case:

We meet some problems with current "settings" when fly close-to-the-ground missions because rangefinder data goes to WPNAV with relatively large lag (because of filter). (BTW in logs the unfiltered data is saved so i spent some time to understand the reason of problem :) )

Our common case is to fly to survey area at 5-10-20 meters above the ground and then make descend to 1 meter above the ground (in global terrain frame). When drone descends with standard 1.5 m\s - there could be the problems. If we "set" RANGEFINDER_WPNAV_FILT_HZ to 0.8 - results are much better. 

Here is SITL example of two equal lights with different RANGEFINDER_WPNAV_FILT_HZ 

[image]

Of cause we can handle it by setting descend speed to some small value, but i think adding RANGEFINDER_WPNAV_FILT_HZ as a parameter will make overall rangefinder process more clear for interested users.

Best regards, 
Mike