ArduPilot/ardupilot #14205: VRUBrain-v51 overflowing flash

peterbarker, 2020-04-27:
## Bug report

**Issue details**

```
_Please describe the problem_
[598/601] Linking build/VRUBrain-v51/bin/arducopter
/home/pbarker/gcc-arm-none-eabi-6-2017-q2-update/bin/../lib/gcc/arm-none-eabi/6.3.1/../../../../arm-none-eabi/bin/ld: bin/arducopter section `.text' will not fit in region `flash'
/home/pbarker/gcc-arm-none-eabi-6-2017-q2-update/bin/../lib/gcc/arm-none-eabi/6.3.1/../../../../arm-none-eabi/bin/ld: region `flash' overflowed by 18572 bytes
collect2: error: ld returned 1 exit status
```

**Version**
master

**Platform**
[  ] All
[  ] AntennaTracker
[ X ] Copter
[ X ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
Plane, Heli, Copter

**Hardware type**
VRUBrain-v51

**Logs**
n/a

peterbarker, 2020-11-25:
No longer overflowing.

closed 2020-11-25 by peterbarker

gaojulong, 2022-07-12:
请问怎么解决的？

rmackay9, 2022-07-12:
@gaojulong,

In case it helps, I did a [blog post here](https://discuss.ardupilot.org/t/ardupilot-pull-request-flash-size-check/86849) which shows how to check if the flash size is overflowing and included a PR that reduced the flash size.