ArduPilot/ardupilot #14210: Copter:  Loading of parameters not restoring everything automatically

mebj, 2020-04-27:
## Bug report

**Issue details**

I have several times reinstalled the 4.0.3 firmware followed by loading of saved parameters and experienced that everything is not reloaded from the backup.

For example the power monitor settings are not recovered automatically. To get those numbers back I have to first enable this function and then it will be recovered. Maybe it is working the same way with other optional hardware but I have not tested this.

This seems to be the case with some parameters also. The example I have so far found is BRD_SAFETYENABLE which after a firmware reinstallation + reload of saved parameters always is 1 even though it should have been saved as 0.

**Version**
Arducopter 4.0.3

**Platform**
[  ] All
[  ] AntennaTracker
[ x ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
F450 clone frame

**Hardware type**
Pixracer R14 Chinese make

**Logs**
_Please provide a link to any relevant logs that show the issue_

rmackay9, 2020-04-27:
The cause of the issue is that some of the parameters are not visible until the enable parameter has been set to a non-zero value.  I can't immediately think of a solution and it might be better to tackle this on the ground station side.

khancyr, 2020-04-27:
yep, that isn't really a bug. Even in the autotest suit we a pushing the parameter list until all are accepted to cope with the parameter tree that need some enabling

mebj, 2020-04-27:
I tried to find in the docs that one should load the params two-three times to be sure but did not see it mentioned. Maybe a note about this is a good first step? This is not a major problem if it is easy to find the proper procedure.
The  BRD_SAFETYENABLE issue may be a Pixracer specific issue. I had no safety switch installed (often not used for Pixracers) and then ESC calibration just failed waiting on the switch as the parameter was set to 1.

SergeyBokhantsev, 2020-04-29:
BTW https://github.com/ArduPilot/MissionPlanner/issues/2068

mebj, 2020-04-30:
In addition to what is mentioned above I have twice or three times experienced that restoring params from backup does not result in a working system. Here is an example from yesterday.

1. Saved params after an autotune session. Drone did fly OK and no indication of any problems.
2. Made an onboard calibration of the compass. Saved the result but later found it was not OK so a restore was needed.
3. Restored the params from step 1. ( used the "three-step" procedure) and rebooted and tried to arm.
4. Was impossible to arm the drone due to "Bad AHRS"
5. Made a total reset and again restored params. Still "Bad AHRS"

So restoring of params from a working drone does not result in that same drone can fly with those parameters. If that is caused by faulthy saving of the params or restoring of them (or Missionplanner vs Arducopter)  I have no idea. All times this has happened to me the "Bad AHRS" has been the resulting fault indication.

Recalibration of accelerometers and compass made the "Bad AHRS" fault disappear.

IamPete1, 2021-08-16:
We have some instructions on how to restore the calibration params, 

https://ardupilot.org/copter/docs/common-parameter-reset.html#option-3-load-another-vehicle-s-firmware

Other that that it is just a matter of setting a few times to un-hide all the enable params. 

I don't think this is a AP issue, maybe we could document better but that should be a wiki issue.

closed 2021-08-16 by IamPete1