ArduPilot/ardupilot #14218: RCx_Option, RTL and Auto do not work properly

lorbass, 2020-04-28:
#13582 
When using RTL or Auto with the RCx_Option depending on the selected Modes according the Mode channel, sometime RTL or Auto works correct, sometimes not or it apears another Mode.
e.g. Flying with FBWB and switched Channel 7 to ppm2000, set for Option RTL, it reports via MavLink to MP Mode "FBWA"
In case of emergency not verry helpful.

_Please describe the problem_

**Version**
ArduPlane 4.0.5 Official

**Platform**
[  ] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
_What type of airframe (flying wing, glider, hex, Y6, octa etc)_

**Hardware type**
Pixhawk 1

**Logs**
_Please provide a link to any relevant logs that show the issue_

peterbarker, 2020-04-29:
Logs?

lorbass, 2020-05-01:
I made a workbench test of different Flight Modes.
RC7 is Auto, RC8 is RTL.
The Log is here: https://drive.google.com/open?id=1-b6KNTR461sg7b6Ww6QCNf9XtqWh-qky
As example at the beginning of the log selecting Auto as RC7 Option from QStabilize. One time correct, next wrong. Seems to be a timing issue. Not recognized with FBWA or FBWB as base.
[image]

peterbarker, 2020-05-02:
Why is your main `FLTMODE_CH` channel and your `RCIN.C7` moving in concert?  I can see how you could get race conditions here:

[image]

lorbass, 2020-05-05:
Hello Peter, thanks for the help. You checked the Log really careful and hit the nail. It was a bad setup in my Taranis TX from a early version. Now it works perfectly. Therefore no ArduPlane Issue. I will close the thread. Otto

closed 2020-05-05 by lorbass