ArduPilot/ardupilot #14224: Copter: altitude jump when EKF core switches

rmackay9, 2020-04-29:
Copter handles EKF resets of yaw and horizontal position but it does not appear to handle resets of altitude.  Below is a screenshot of a real flight in which the EKF2 cores were switched using an auxiliary switch ([code is here](https://github.com/rmackay9/rmackay9-ardupilot/commits/ekf-lane-switch-bool)).

[image]

rmackay9, 2021-07-14:
This issue came up again during Copter-4.1.0-beta testing: https://discuss.ardupilot.org/t/7inch-quadcopter-ac-4-1-0-beta5-inflight-exk3-lane-switch-root-cause-analysis-request/73507/6

RickReeser, 2024-06-10:
[I have encountered this](https://discuss.ardupilot.org/t/ekf-lane-change-causes-change-in-canonical-altitude/116544) a couple times. In my case, I was controlling the drone in Guided using the altitude reported by GLOBAL_POSITION_INT. This lane change causes that altitude to be **_very_** wrong in some cases, which becomes a safety issue if it is being used to control the drone.