ArduPilot/ardupilot #14229: New builds with EKF3 will not pass pre-arm on OmnibusF4pro and MatekF405Wing

Hwurzburg, 2020-04-29:
With the change to remove EKF2 on 1M boards, my builds on MatekF405-Wing and OmnibusF4pro no longer pass pre-arm checks. GPS velocity errors prevent EKF3 to set origin and start using GPS.
These are MN8 GPS, BN220 and BN880. Previously with EKF2 the autopilot would get a passing EKF and AHRS in a minute or so , soon after acquiring 8-9 sats and HDOP <1.5....

using the same firmware (a recent master plus some PRs for bidir Frsky and tailsitter QASSIST mode), but just building EKF2 and using the exact same hardware, ground location, parameters, etc. it passes pre-arm checks allowing arming quickly, with just the change of only building and using EKF3, it will not pass (at least within 6-7minutes and high sat/low HDOP)
 logs showing this with only the change of EKFS are:

EKF3 failure: https://www.dropbox.com/s/0nq7wyt22hsh35d/2020-04-28%2015-07-30.bin?dl=0
and same using EKF2 https://www.dropbox.com/s/hwwdoqd9z0wrvm2/log_1_2020-4-29-09-06-00.bin?dl=0
**Platform**
[  ] All
[  ] AntennaTracker
[  ] Copter
[ x ] Plane b
[  ] Rover
[  ] Submarine

vierfuffzig, 2020-05-02:
@Hwurzburg your GPS seems to be a 5 / 6th gen really (hw 00040007 on sw 7.03). your log on ek3 shows a noticeable speed spike of 110 m/s initially. the ek2 log does not show any gps speed spikes above 1 m/s.
i just flashed one of my setups with latest on a matekf405-wing (genuine m8n) and tried arming in the backyard - no noticeable differences on ek3 vs ek2. here's the log: https://github.com/vierfuffzig/Chibios-miscellaneous/blob/master/00000012.BIN
i'd say it was that initial glitch followed by a marginal fix that made your ek3 unhappy. someone more knowledgeable might want to chime in though.
cheers, basti

Hwurzburg, 2020-05-02:
thanks, that GPS was an older version, yes, but new BN-220 and BN-880 MN8 gps do the same..Tridge says that its the fact that they are mini sized, and that the EKF scaling is different between EKF2 and EKF3 on plane...EKF2 is  50% more lax in plane than EKF3...(I will do a PR to fix that for plane, but not for copter where GPS vel is more critical)... I will also add some notes in the Wiki about how to cope if it happens (it probably will with minis in copters, although they mostly use full size on poles)...also can happen with noise from RF onboard Tridge says....
solutions are to wait (takes about 15min on my new minis...to get enough sats 12+ and settling time)....relax the scaling param (not recommended on copters)...or possibly increase the ground plane under the GPS with AL foil (going to try that one for sure)...

closed 2020-05-02 by Hwurzburg