ArduPilot/ardupilot #14232: constrain_nan triggered in LowPassFilter

peterbarker, 2020-04-30:
## Bug report

**Issue details**

This call to constrain_float is causing the `is_nan` check to fail:
https://github.com/ardupilot/ardupilot/blob/master/libraries/Filter/LowPassFilter.cpp#L28

It is entirely possible `dt` is wonky in there - this is associated with a significant outage in telemetry from the hardware.

**Version**
master

**Platform**
[  ] All
[  ] AntennaTracker
[ X ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
None / benchtest

**Hardware type**
Cube on standard carrier board with a telemetry radio on telem1

**Logs**

way too large to transfer.

peterbarker, 2020-04-30:
@andyp1per FFT stuff is the new kid on the block for this.  Any way we could be passing in bad data - particularly timestamps - to a LPF?

andyp1per, 2020-04-30:
@peterbarker for FFT data - yes certainly, the algorithm will return nonsense in certain circumstances which I have to remove. Timestamps, not so sure - do you know where this is getting called from?

peterbarker, 2020-04-30:
On Wed, 29 Apr 2020, Andy Piper wrote:

> @peterbarker for FFT data - yes certainly, the algorithm will return nonsense in certain circumstances which I have to remove. Timestamps, not so sure -
> do you know where this is getting called from?

This is as far back as I've traced it.  I have an inkling that if you were 
to insert a delay into the main loop of ~3 seconds you'd be able to 
reproduce.  I'm only *guessing* FFT as it's the new kid on the block - 
sorry!

peterbarker, 2020-11-25:
This code has now been modified.  We'll have to attack this one more vigorously if it happens again.

closed 2020-11-25 by peterbarker