ArduPilot/ardupilot #14245: AP_Scripting: Broken SITL build on OSX

jmachuca77, 2020-04-30:
## Bug report
Current master is not building on OSX due to issues with AP_Scripting. Build log attached below.

**Issue details**
```console
$ ./waf rover
Waf: Entering directory `/Users/jaime/Developer/AION/ardupilot_private/build/sitl'
[4/4] Creating build/sitl/ap_version.h
[169/634] Compiling libraries/AP_Logger/LoggerMessageWriter.cpp
[190/634] Compiling libraries/AP_Mission/AP_Mission_Commands.cpp
[469/634] Compiling libraries/SITL/SIM_ICEngine.cpp
[501/634] Compiling libraries/SITL/SIM_Rover.cpp
[515/634] Compiling libraries/AP_Scripting/lua_scripts.cpp
[516/634] Compiling libraries/AP_Scripting/lua_repl.cpp
[517/634] Compiling libraries/AP_Scripting/lua_bindings.cpp
[565/634] Compiling libraries/AP_BattMonitor/AP_BattMonitor.cpp
[566/634] Compiling libraries/AP_BattMonitor/AP_BattMonitor_Params.cpp
[567/634] Compiling libraries/AP_BoardConfig/canbus_driver.cpp
[568/634] Compiling libraries/AP_BoardConfig/AP_BoardConfig.cpp
[569/634] Compiling libraries/AP_BoardConfig/AP_BoardConfig_CAN.cpp
[570/634] Compiling libraries/AP_InertialSensor/AP_InertialSensor.cpp
[571/634] Compiling libraries/AP_Mission/AP_Mission.cpp
In file included from ../../libraries/AP_Scripting/lua_repl.cpp:7:
In file included from ../../libraries/AP_Scripting/lua_generated_bindings.h:13:
In file included from ../../libraries/AP_Vehicle/AP_Vehicle.h:33:
In file included from ../../libraries/AP_Relay/AP_Relay.h:13:
In file included from ../../libraries/AP_HAL/I2CDevice.h:20:
In file included from /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/../include/c++/v1/vector:274:
In file included from /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/../include/c++/v1/__bit_reference:15:
/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/../include/c++/v1/algorithm:2067:34: fatal error: too many arguments provided to function-like macro invocation
remove(_ForwardIterator __first, _ForwardIterator __last, const _Tp& __value_)
                                 ^
../../libraries/AP_Filesystem/posix_compat.h:91:9: note: macro 'remove' defined here
#define remove(pathname) apfs_remove(pathname)
        ^
1 error generated.

In file included from ../../libraries/AP_Scripting/lua_scripts.cpp:21:
In file included from ../../libraries/AP_Scripting/lua_generated_bindings.h:13:
In file included from ../../libraries/AP_Vehicle/AP_Vehicle.h:33:
In file included from ../../libraries/AP_Relay/AP_Relay.h:13:
In file included from ../../libraries/AP_HAL/I2CDevice.h:20:
In file included from /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/../include/c++/v1/vector:274:
In file included from /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/../include/c++/v1/__bit_reference:15:
/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/../include/c++/v1/algorithm:2067:34: fatal error: too many arguments provided to function-like macro invocation
remove(_ForwardIterator __first, _ForwardIterator __last, const _Tp& __value_)
                                 ^
../../libraries/AP_Filesystem/posix_compat.h:91:9: note: macro 'remove' defined here
#define remove(pathname) apfs_remove(pathname)
        ^
1 error generated.

In file included from ../../libraries/AP_Scripting/lua_bindings.cpp:8:
In file included from ../../libraries/AP_Scripting/lua_generated_bindings.h:13:
In file included from ../../libraries/AP_Vehicle/AP_Vehicle.h:33:
In file included from ../../libraries/AP_Relay/AP_Relay.h:13:
In file included from ../../libraries/AP_HAL/I2CDevice.h:20:
In file included from /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/../include/c++/v1/vector:274:
In file included from /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/../include/c++/v1/__bit_reference:15:
/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/../include/c++/v1/algorithm:2067:34: fatal error: too many arguments provided to function-like macro invocation
remove(_ForwardIterator __first, _ForwardIterator __last, const _Tp& __value_)
                                 ^
../../libraries/AP_Filesystem/posix_compat.h:91:9: note: macro 'remove' defined here
#define remove(pathname) apfs_remove(pathname)
        ^
1 error generated.

Waf: Leaving directory `/Users/jaime/Developer/AION/ardupilot_private/build/sitl'
Build failed
 -> task in 'objs/AP_Scripting' failed (exit status 1): 
	{task 4558346168: cxx lua_repl.cpp -> lua_repl.cpp.0.o}
 (run with -v to display more information)
 -> task in 'objs/AP_Scripting' failed (exit status 1): 
	{task 4558346280: cxx lua_scripts.cpp -> lua_scripts.cpp.0.o}
 (run with -v to display more information)
 -> task in 'objs/AP_Scripting' failed (exit status 1): 
	{task 4558345832: cxx lua_bindings.cpp -> lua_bindings.cpp.0.o}
 (run with -v to display more information)
```

**Version**
Master

**Platform**
[X] All

jmachuca77, 2020-05-06:
Checked on master and this issue is not present there.

closed 2020-05-06 by jmachuca77

gamoreno, 2021-04-19:
I just cloned the repo and got the same error. This in in macOS Catalina.
I just installed the prereqs for Mac using the script in [this commit](https://github.com/ArduPilot/ardupilot/commit/23c8ef1c3ef668e64b779a3d9701909f4bcb83e0)
(the original wasn't working for me)

This build was launched by invoking
```
sim_vehicle.py -j4 -f quadplane --console --map -v ArduPlane -L KSFO
```

Am I doing something wrong?

```
[641/816] Compiling libraries/AP_Scripting/lua/src/ldblib.c
In file included from ../../libraries/AP_Scripting/lua_scripts.cpp:16:
In file included from ../../libraries/AP_Scripting/lua_scripts.h:23:
In file included from ../../libraries/AP_Scripting/AP_Scripting.h:23:
In file included from ../../libraries/AP_HAL/I2CDevice.h:20:
In file included from /Library/Developer/CommandLineTools/usr/bin/../include/c++/v1/vector:274:
In file included from /Library/Developer/CommandLineTools/usr/bin/../include/c++/v1/__bit_reference:15:
/Library/Developer/CommandLineTools/usr/bin/../include/c++/v1/algorithm:2095:34: fatal error: too many arguments provided to function-like macro invocation
remove(_ForwardIterator __first, _ForwardIterator __last, const _Tp& __value_)
                                 ^
../../libraries/AP_Filesystem/posix_compat.h:91:9: note: macro 'remove' defined here
#define remove(pathname) apfs_remove(pathname)
        ^
1 error generated.

In file included from ../../libraries/AP_Scripting/lua_bindings.cpp:9:
In file included from libraries/AP_Scripting/lua_generated_bindings.h:4:
In file included from ../../libraries/AP_HAL/I2CDevice.h:20:
In file included from /Library/Developer/CommandLineTools/usr/bin/../include/c++/v1/vector:274:
In file included from /Library/Developer/CommandLineTools/usr/bin/../include/c++/v1/__bit_reference:15:
/Library/Developer/CommandLineTools/usr/bin/../include/c++/v1/algorithm:2095:34: fatal error: too many arguments provided to function-like macro invocation
remove(_ForwardIterator __first, _ForwardIterator __last, const _Tp& __value_)
                                 ^
../../libraries/AP_Filesystem/posix_compat.h:91:9: note: macro 'remove' defined here
#define remove(pathname) apfs_remove(pathname)
        ^
1 error generated.

In file included from ../../libraries/AP_Scripting/lua_repl.cpp:6:
In file included from ../../libraries/AP_Scripting/lua_scripts.h:23:
In file included from ../../libraries/AP_Scripting/AP_Scripting.h:23:
In file included from ../../libraries/AP_HAL/I2CDevice.h:20:
In file included from /Library/Developer/CommandLineTools/usr/bin/../include/c++/v1/vector:274:
In file included from /Library/Developer/CommandLineTools/usr/bin/../include/c++/v1/__bit_reference:15:
/Library/Developer/CommandLineTools/usr/bin/../include/c++/v1/algorithm:2095:34: fatal error: too many arguments provided to function-like macro invocation
remove(_ForwardIterator __first, _ForwardIterator __last, const _Tp& __value_)
                                 ^
../../libraries/AP_Filesystem/posix_compat.h:91:9: note: macro 'remove' defined here
#define remove(pathname) apfs_remove(pathname)
        ^
1 error generated.

Waf: Leaving directory `/Users/gmoreno/swdev/ardupilot/build/sitl'
Build failed
 -> task in 'objs/AP_Scripting' failed (exit status 1): 
	{task 4374658960: cxx lua_scripts.cpp -> lua_scripts.cpp.0.o}
 (run with -v to display more information)
 -> task in 'objs/AP_Scripting' failed (exit status 1): 
	{task 4374658624: cxx lua_bindings.cpp -> lua_bindings.cpp.0.o}
 (run with -v to display more information)
 -> task in 'objs/AP_Scripting' failed (exit status 1): 
	{task 4374658848: cxx lua_repl.cpp -> lua_repl.cpp.0.o}
 (run with -v to display more information)
SIM_VEHICLE: Build failed
SIM_VEHICLE: Killing tasks
```