ArduPilot/ardupilot #14252: ArduPlane: Parachute release does not stop quad motors

Davidsastresas, 2020-05-01:
**Platform**
[  ] All
[  ] AntennaTracker
[  ] Copter
[ x ] Plane
[  ] Rover
[  ] Submarine

Parachute release only stops forward flight motor, but not quad motors. There are some conditions that override the trottle_suppressed flag condition in update_throttle_suppression() in quadplane.cpp, so a parachute release does not stop quad motors.

peterbarker, 2024-09-22:
@Davidsastresas this looks like it could still be a problem.

Are you planning on pushing forward with this?

peterbarker, 2024-09-22:
@Hwurzburg you worked on something similar to this, https://github.com/ArduPilot/ardupilot/pull/24532  

This PR would cover the case where you pop a chute manually but are still armed.

It would also *seem* to be a superset of the functionality in the PR I mentioned above.

Davidsastresas, 2024-09-22:
You are right, thanks for mentioning me. I think we can close this one.

closed 2024-09-22 by Davidsastresas