ArduPilot/ardupilot #14253: EKF2/3: GPS glitch can cause emergency yaw reset

rmackay9, 2020-05-01:
I've found that during SITL testing of both EKF2 and EKF3 that a GPS glitch can cause an emergency yaw reset to occur which then leads to the vehicle no longer using the compass.

To recreate the issue do this:

- set all parameters to defaults (i.e. rm eeprom.bin)
- start sim_vehicle from the copter directory (cd ArduCopter; ../Tools/autotest/sim_vehicle.py --map --console)
- arm throttle
- rc 3 1800 (to climb to 10m)
- rc 3 1500 (to stop climbing)
- rc 2 1200 (to start the vehicle flying forward)
- param set SIM_GPS_GLITCH_Y 0.0001

[image]

peterbarker, 2020-05-02:
I can also be triggered with really vibes.

```
mode loiter
param set SIM_ACC_RND 200
param set SIM_ACC_RND 200
arm throttle
rc 3 2000
```

[image]

20m/s/s isn't healthy, but... disabling mag sensors?

priseborough, 2020-05-02:
I will add a check for these sensor failure modes where possible and inhibit the yaw reset request if it is likely that the impending EKF failsafe is caused by something other than bad yaw.

The other idea is to limit the yaw reset to the first 30 seconds after takeoff. If the EKF failsafe hasn't started triggering before then, then it is highly likely the yaw angle obtained form the mag is good enough. That is the strategy adopted by PX4.

priseborough, 2020-05-02:
> 20m/s/s isn't healthy, but... disabling mag sensors?

If a yaw reset was necessary due to a bad yaw angle, then it needs to be assumed that the magnetic yaw was unreliable because using a bad magnetometer after a reset eventually results in the yaw rotating back to a bad value with another loss of navigation.

The underlying issue here is to that we need to inhibit the yaw reset request if there is an error indication in sensors other than the magnetometer.

tridge, 2020-05-03:
I just discovered that the GSF changes also broke EK3_MAG_CAL=5 with GPS yaw. It never gets yaw alignment.
To reproduce:
 - AHRS_EKF_TYPE=3
 - EK3_ENABLE=1
 - EK3_MAG_CAL=5
 - SIM_GPS_TYPE=5
 - SIM_GPS_HDG=1
 - GPS_TYPE=5

A git bisect shows it broke at:
AP_NavEKF3: Fix integration of GPS yaw options
863f9891301d7b55f3c8ee35e6474847904a4e8b

tridge, 2020-05-03:
fix here: https://github.com/ArduPilot/ardupilot/pull/14266

priseborough, 2020-05-03:
The reset in both these cases is being triggered internally within the EKF and not by the external EKF failsafe. I have removed the internal trigger mechanism verified for both EKF2 and EKF3 that:

A simulated GPS positon jump does not trigger a yaw reset
A simulated IMU clipping does not trigger a yaw reset
A simulated flyaway induced by setting COMPASS_AUTODEC = 0 and COMPASS_DEC = -3 reults ina yaw reset and recovery.

A PR will be submitted shortly.

priseborough, 2020-05-03:
Something I did notice during testing is that EKF3 sometimes performs a lane switch when there is a large GPS glitch, whereas EKF2 does not. This is something to investigate.

priseborough, 2020-05-03:
Fix here: https://github.com/ArduPilot/ardupilot/pull/14275

closed 2020-05-14 by tridge