ArduPilot/ardupilot #14260: Cascaded ALT configs only partially working

Hwurzburg, 2020-05-02:
Having recently added a fourth board config to the OminbusF4pro to allow full access to UART4 as the BRD_ALT_CONFIG =3, and tested it sucessfully (#14251) , I wanted to add another that combined Config 1 (USART3 instead of I2C) and 3 (UART4 instead of RSSI and OUTPUT 5) and was surprised that the new option only activated USART3 and not UART4...
here are the  portions of the file dealing with redefinition:
............
#analog rssi pin (also could be used as analog airspeed input)
PA0 RSSI_IN ADC1
define BOARD_RSSI_ANA_PIN 0

#pwm output. 1 - 4 on main header, 5 & 6 on separated header w/o 5V supply, 7 & 8 on CH5 and CH6 pads
PB0 TIM1_CH2N TIM1 PWM(1) GPIO(50)
PB1 TIM1_CH3N TIM1 PWM(2) GPIO(51)
PA3 TIM2_CH4  TIM2 PWM(3) GPIO(52)
PA2 TIM2_CH3  TIM2 PWM(4) GPIO(53)
PA1 TIM2_CH2  TIM2 PWM(5) GPIO(54)
PA8 TIM1_CH1  TIM1 PWM(6) GPIO(55)
PC8 TIM8_CH3  TIM8 PWM(7) GPIO(56)
PC9 TIM8_CH4  TIM8 PWM(8) GPIO(59)

# use I2C pins as USART3 (SERIAL2) in BRD_ALT_CONFIG= 1 & 4
PB10 USART3_TX USART3 ALT(1)
PB11 USART3_RX USART3 ALT(1)
PB10 USART3_TX USART3 ALT(4)
PB11 USART3_RX USART3 ALT(4)

# UART4 TX available as an alternative config on PA0 (RSSI pad) with BRD_ALT_CONFIG= 2
PA0 UART4_TX UART4 ALT(2)

# full UART4 also available as alt config on PA0 (RSSI pad) and PA1 (PWM output chan 5) with BRD_ALT_CONFIG= 3 & 4
PA0 UART4_TX UART4 ALT(3)
PA1 UART4_RX UART4 ALT(3)
PA0 UART4_TX UART4 ALT(4)
PA1 UART4_RX UART4 ALT(4)

the first cascaded redefine for USART3 works...but UART 4 works only in Config3 not 4

**Platform**
[x  ] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
_What type of airframe (flying wing, glider, hex, Y6, octa etc)_

**Hardware type**
_What autopilot hardware was used? (Pixhawk, Cube, Pixracer, Navio2, etc)_

**Logs**
_Please provide a link to any relevant logs that show the issue_

Hwurzburg, 2020-05-06:
here is a link to the compiled hwdef.h file that results.... https://www.dropbox.com/s/hdequ1pw78rs82e/hwdef.h?dl=0

Hwurzburg, 2020-05-08:
@vierfuffzig Basti, do you have time to look at this?

vierfuffzig, 2020-05-09:
@Hwurzburg i‘ll try. i too would love to get a lot more out of alt configs, it‘s just that the hwdef scripting‘s complexity has scared me off...

Hwurzburg, 2020-05-10:
actually is working.....dont know what happened in first bench tests....probably a port miss-configuration

closed 2020-05-10 by Hwurzburg