ArduPilot/ardupilot #14267: sim_vehicle.py fails to compile with --osd flag enabled on Ubuntu 18.04

npyoung, 2020-05-03:
## Bug report

**Issue details**

After configuring and building ArduPlane locally (for SITL) on Ubuntu 18.04, I get a compilation failure when I run `sim_vehicle.py` with `--osd`. Looks like something's trying to compare incompatible integers.

```
[147/669] Compiling libraries/AP_Gripper/AP_Gripper_EPM.cpp
In file included from …/…/libraries/AP_Filesystem/AP_Filesystem_ROMFS.cpp:21:0:
…/…/libraries/AP_Math/AP_Math.h: In instantiation of ‘decltype (((one < two) ? one : two)) MIN(const A&, const B&) [with A = unsigned int; B = int; decltype (((one < two) ? one : two)) = unsigned int]’:
…/…/libraries/AP_Filesystem/AP_Filesystem_ROMFS.cpp:100:49: required from here
…/…/libraries/AP_Math/AP_Math.h:209:16: error: comparison between signed and unsigned integer expressions [-Werror=sign-compare]
return one < two ? one : two;
^
compilation terminated due to -Wfatal-errors.
cc1plus: some warnings being treated as errors

Waf: Leaving directory `/home/noah/code/ardupilot/build/sitl’
Build failed
-> task in ‘objs/AP_Filesystem’ failed (exit status 1):
{task 140387936722512: cxx AP_Filesystem_ROMFS.cpp -> AP_Filesystem_ROMFS.cpp.0.o}
(run with -v to display more information)
SIM_VEHICLE: Build failed
```

See original report in the forums [here](https://discuss.ardupilot.org/t/arduplane-sitl-build-failing-on-ubuntu-18-04/55804).

**Version**

Ardupilot master branch commit 29d8586
Ubuntu 18.04

**Platform**
[  ] All
[  ] AntennaTracker
[  ] Copter
[x] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
plane

**Hardware type**
SITL

peterbarker, 2020-05-03:
@npyoung any chance you could test https://github.com/ArduPilot/ardupilot/pull/14268 ?

closed 2020-05-04 by tridge

npyoung, 2020-05-04:
Now working :)

PR #14268 (2020-05-04): AP_Filesystem: correct comparison of signed/unsigned
Closes #14267

... also, don't seek to an absolute negative offset, you won't like it.
commit f43e48a6 (2020-05-04): AP_Filesystem: correct comparison of signed/unsigned

Closes #14267