ArduPilot/ardupilot #14283: Copter/Failsafe: GCS_ENABLE and THR_ENABLE

lightbulb14, 2020-05-04:
This is not really a bug, but a personal preference. The meanings listed for FS_GCS_ENABLE and FS_THR_ENABLE have the same choices, but in different orders. Would it be possible to change the FS_THR_ENABLE values so that the same value corresponds the same action? ie currently FS_GCS_ENABLE==3 and FS_THR_ENABLE==4 both are described as "Enabled always SmartRTL or RTL". I was initially confused since options 0, 1, and 2 are identical so I assumed that options 3, 4 and 5 were identical as well which caused unexpected behavior in the SIM.

Suggested FS_THR_ENABLE values:
Value | Meaning
-- | --
0 | Disabled
1 | Enabled always RTL
2 | Enabled Continue with Mission in Auto Mode (Deprecated in 4.0+)
3 | Enabled always SmartRTL or RTL
4 | Enabled always SmartRTL or Land
5 | Enabled always land (4.0+ Only)

Found using Mission Planner (1.3.70) built in simulator with Arducopter 4.0.4-dev
Thanks, Lightbulb14

rmackay9, 2020-05-04:
Hi @lightbulb14, I wish we could but it would unfortunately break existing setups.  As we move to consolidate failsafes across the vehicles we may be able to consolidate them some more.

Pedals2Paddles, 2020-05-05:
I'm glad I'm not the only one annoyed by that.  When I was doing the big failsafe refactoring last fall, that is the one thing I couldn't redo that drove me absolutely bonkers.  But ya i would wreak havoc on existing vehicle configurations and GCS applications so has to stay the way it is until the whole thing is unified.

closed 2020-05-05 by Pedals2Paddles