ArduPilot/ardupilot #14286: Quadplane allows arming in STABILIZE mode

jschall, 2020-05-04:
# Bug report

**Issue details**

_Please describe the problem_
I had accidentally selected STABILIZE mode, thinking that it was Q_STABILIZE. The vehicle successfully armed, started the quadplane motors, and drifted around about 20cm above the ground, with none of the controls functioning. It drifted into some grass and I force disarmed it via GCS. This is an easy mistake to make and is a safety issue that could potentially have hurt a person.

**Version**
_What version was the issue encountered with_
around bd0ebb5778211c7fff4d35305b3b3866151e0bb0
**Platform**
[  ] All
[  ] AntennaTracker
[  ] Copter
[ x ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
_What type of airframe (flying wing, glider, hex, Y6, octa etc)_
Quadplane
**Hardware type**
_What autopilot hardware was used? (Pixhawk, Cube, Pixracer, Navio2, etc)_
Cube Black

**Logs**
_Please provide a link to any relevant logs that show the issue_
https://drive.google.com/file/d/101pqduUAPaeccEIGPtBjjxGwgjtAzNgS/view?usp=sharing

IamPete1, 2020-05-05:
looks like you instantly went into Qasist trying to maintain constant altitude. I think its probably fair to refuse to arm in a plane mode if Qassist will be active.

WickedShell, 2020-05-05:
@IamPete1 it can't just be when Qassist will be active I don't think. I regularly arm quadplanes in auto while on the ground doing a takeoff, and that works perfectly. By definition QAssist would want to be active then but can't be. We wouldn't want to block that use case.

Personally I'm not actually a fan of refusing to arm while in a given mode, if requested we should allow it, but I may need to give it more thought.

jschall, 2020-05-05:
Well, there is definitely no reason why the vehicle should arm into a state where it will be uncontrollable and drift into things...

WickedShell, 2020-05-05:
Technically not uncontrollable right? Just needed you to raise the forward throttle and get it flying :) But yes in practice I agree what you actually got here wasn't good.

IamPete1, 2021-08-28:
Fixed by https://github.com/ArduPilot/ardupilot/pull/18413, set `Q_OPTION` bit 18 to prevent arming in forward flight modes

closed 2021-08-28 by IamPete1