ArduPilot/ardupilot #14288: EKF3: "still initialising" or "EKF attitude is bad" if compass disabled while using external nav

rmackay9, 2020-05-05:
If the EKF3 is using "external navigation" for yaw, the vehicle cannot be armed because the "EKF3 still initailising" pre-arm check is triggered.

To reproduce the issue in SITL:

- param set EK2_ENABLE 0
- param set EK3_ENABLE 1
- param fetch
- param set EK3_GPS_TYPE 3
- param set EK3_MAG_CAL 5 ("Use external yaw sensor")
- param set AHRS_EKF_TYPE 3
- param set VISO_TYPE 1
- param set COMPASS_USE 0
- param set COMPASS_USE2 0
- param set COMPASS_US3 0
- param set SERIAL3_PROTOCOL 1

Reboot SITL and start with vicon enabled:

- ../Tools/autotest/sim_vehicle.py --map --console -A "--uartB=sim:vicon:"

Right-mouse-click on the MAVProxy map and select, "Set Origin (with height)".  At this point normally the vehicle would appear on the map but it won't.

The vehicle will appear on the map if the compass is enabled as the EKF's yaw source:

- param set COMPASS_USE 1
- param set EK3_MAG_CAL 3 ("After first climb yaw reset")

Below is a screen shot from SITL
[image]

Thanks very much to @anbello for testing and uncovering this issue

anbello, 2020-05-08:
It seems to me that @tridge with this commit https://github.com/ArduPilot/ardupilot/commit/0c1ba48212eac9cb871bfdf7993b4fe2e0fa01c3 has solved this issue.
I tested on SITL and I will test on real flight asap.
@rmackay9 can you confirm?

anbello, 2020-05-12:
@rmackay9 I tested with real copter and the commit https://github.com/ArduPilot/ardupilot/commit/0c1ba48212eac9cb871bfdf7993b4fe2e0fa01c3 solve the issue but there are other issue.

With COMPASS_USE 0 and EK3_MAG_CAL 5 when you set origin there are no "EKF3 still initailising" or "EKF attitude is bad" messages and the vehicle appear on the map.

But then the EKF variance become huge with big drift on yaw, pitch and roll visible on the HUD (with vehicle on ground), and the vehicle is unflyable.

With COMPASS_USE 1 and EK3_MAG_CAL 3 is all OK as before.

rmackay9, 2020-05-13:
@anbello, you're right that this issue seems to be fixed in master which is great.  I'll probably close this issue then and we can create a new one for the other issue you've found.

Re the other issue, I can't reproduce any problems in SITL.  So if I set COMPASS_USE = 0 (and all the other compasses to zero as well) and EK3_MAG_CAL=5 then it all seems to work fine.  Do you experience a problem in a real flight?  If yes then do you have log?  I suspect the issue is the yaw from the camera was bad for some reason.

closed 2020-05-13 by rmackay9

rmackay9, 2020-05-13:
@anbello, I have probably recreated the issue you've seen in SITL and raised a new issue https://github.com/ArduPilot/ardupilot/issues/14356.

anbello, 2020-05-13:
> Do you experience a problem in a real flight? If yes then do you have log? I suspect the issue is the yaw from the camera was bad for some reason.

Yes the problem described was in real flight, or better in real copter moved by hand because with such huge EKF compass error was impossible to flight.

I have no log but if you want I can do it with LOG_DISARMED = 1.

Anyway I see you reproduced the issue also in SITL, and yes, also my suspect was for the yaw from external data.

rmackay9, 2020-05-13:
@anbello, so this only happened when the vehicle was disarmed right?

1qkjqdckuqdckj, 2021-09-17:
> If the EKF3 is using "external navigation" for yaw, the vehicle cannot be armed because the "EKF3 still initailising" pre-arm check is triggered.
>
> To reproduce the issue in SITL:
>
> * param set [...]

Hi. Sorry I'm totally a noob. i have some problems with pixhawk px4 board. First after i set everything and try to fly. The quadcopter never stabilizat. Always going to one direction. And after that i got ekf problem. I noticed after i flashed it to 4.0.7 in mission planer. I chose pixhawk 1 like firmware because i didn't find the px4 target or something. I noticed under the connexion tab. It's writing octorotor com 8. I was pretty sure i flashed quadcopter. Now i reflashed it to the same target. But still have that ekf problem. Is there something i can do?
Thanks

1qkjqdckuqdckj, 2021-09-17:
Hi. Sorry i have some problems with px4
 board. First after i set everything and try to fly. The quadcopter never stabilizat. Always going to one direction. And after that i got ekf problem. I noticed after i flashed it to 4.0.7 in mission planer. I chose pixhawk 1 like firmware because i didn't find the px4 target or something. I noticed under the connexion tab. It's writing octorotor com 8. I was pretty sure i flashed quadcopter. Now i reflashed it to the same target. But still have that ekf problem. Is there something i can do?
Thanks