ArduPilot/ardupilot #14293: Plane: Impossible to set SERVO9/10_MAX/MIN/TRIM for BL Heli32 ESC's Out

lorbass, 2020-05-05:
## Bug report
I flashed the FC with the new Version. After that, the motors didn't work correct and during the maden flight the wing made a salto left. One motor stoped and the other got full power.
The reason: The SERVOx_MAX/MIN/TRIM where higher/lower than in the previous Version 3.9.8.

**Issue details**
I tried to set this params via MP to 1100/1900 to the old values and it reported: Parameters succesfully saved.
But after "Refresh Params" it showed again 1000/2000.
I use ESC BLHeli32 with the setup according the Wiki for the SERVO Out 9 and 10.
For me I got around this problem, by recalibrating my ESC's. But for other it could be dangerous because the wrong message: parameters succesfully saved.

**Version**
Plane 4.0.5 Official

**Platform**
[  ] Plane

**Airframe type**
Dual motor vectored Tailsitter Skywalker 6

**Hardware type**
Pixhawk 1

**Logs**
No Log, but you need to rename attached file to .PARAM
[Param setup issue.TXT](https://github.com/ArduPilot/ardupilot/files/4579292/Param.setup.issue.TXT)

lorbass, 2020-05-05:
It's even worse. The RCOU C10 goes up (left wing) nearly to the max. but the Motor does not rotate. When arming it rotates slowly, but when increasing the trottlle it stops. RCOU C9 is OK.
And when shaking the wing in roll, no response to hold the level.
After flashing Plane 3.9.8  every thing works as bevor flashing Plane 4.0.5 Official (also Issue above)
Log of a Rack test: https://drive.google.com/open?id=1o86fmhmU-sBFbl55bbL0XLbv97ByGKaZ

kd0aij, 2020-05-08:
RC7_OPTION=16 causes a panic on boot in SITL:
RC function (16) initialisation not handled

But with RC7_OPTION=0, I am able to set and save SERVO9_MIN and MAX

lorbass, 2020-05-12:
Thanks for the answer. Sorry for the late response, I do not get a notification as author as mentionned. 
You have a good nose to find such hidden issues. Will try it.
And then test again if also the second issue is solved. (wrong RCOU10)

lorbass, 2020-05-12:
I'v flashed Plane 4.0.5 again and set all RCx_Options to Zero. 
But Servo9/10_Min/Max/Trim where consequently reset to 1000/2000/1000 after tuning to 1100/1900/1500 when refreshing Params or disconnect/connect.

kd0aij, 2020-05-12:
I'll try to reproduce this on a PixHawk1. 
Are SERVO1-8_MIN/MAX/TRIM parameters behaving properly on 4.0.5?

lorbass, 2020-05-13:
Yes, Servo1-8 works correct. But I found, how to reproduce this error!

1. I flashed a new FC (Pixhwahk 4 Mini Holybro) with Plane 4.0.5.
2. Then set the Params with the button "Reset to defaults" in case they are set somehow.
3. Set Param Q_ENABLE  = 1 (seems to be needed to reproduce the issue)
4. Check the Servo9/10 Min/Max/Trim if they are still at 1100/1900/1500 as with defaults. Yes OK
5. Now set the Param SERVO_BLH_MASK  = 768 (as for channel 9 and 10)
6. Check Servo9/10 and **BINGO**: The values changed to 1000/2000
7. But now, when setting SERVO_BLH_MASK back to Zero, the error remains until beginning again at item 2.
I found this Param by setting one by one from default to my setting top down until the issue happened via the table in "Compare Params"
There where "several" go arounds needed until "SERVO_BLH_MASK" :-(
This issue is not with V 3.9.8, and I did not test the Versions between.

kd0aij, 2020-05-13:
So you have left/right motors on servo 9/10, with blheli escs set to dshot150, correct?
The code here: https://github.com/ArduPilot/ardupilot/blob/master/libraries/SRV_Channel/SRV_Channel_aux.cpp#L184
forces min/max/trim to 1000/2000/(1000 || 1500) if the output type is Dshot, but it hasn't changed in 2 years.  I would guess that it has been in the stable releases for a while.

lorbass, 2020-05-14:
> So you have left/right motors on servo 9/10, with blheli escs set to dshot150, correct?

Yes, due to get ESC Telemetry

The Min/Max/Trim Issue, tested with above procedure, does not happen with V 3.9.8- V3.9.9
From V 4.0.0 up to V 4.0.5 the Min/Max/Trim Values are fixed to 1000/2000/1000.

But this is not the main issue as described in my second post in this thread.
There must be something wrong with this type/Nr of Output, even with V 3.9.8. When you have seen my last Test, the RCOU9 suddenly stucked at 1500 for some seconds. 
Video: https://youtu.be/c6VeOfRRTCU
Log: https://drive.google.com/open?id=1hZ9xyqhZn-4UPspAgZqbSEVSiItq4OFd
[image]

kd0aij, 2020-05-14:
The link to log above is wrong.  I assume that wasn't a mode change at line 543000?

lorbass, 2020-05-14:
Sorry, this here is the right Log:
https://drive.google.com/open?id=1jzTWXbyMloUfPloC_LcMQe4dSTwS8vHZ
No, not a mode change. QHover with Pitch forward up to 65° and it happened at the end of pitch back to vertical. I set Servo9/10 Trim to 1500. Is that correct or should it be like Min(1100)? I ask, because this "lock down" shows just 1500.

kd0aij, 2020-05-14:
This is interesting: the log value for DesYaw is often zero (implying non-VTOL mode) and it becomes non-zero at the beginning of the uncommanded yaw event:
[image]
@IamPete1 It looks to me as though QASSIST is disabled by the parameters, but could this have something to do with it?

lorbass, 2020-05-14:
I'm not shure if DesYaw Missing has to do with this issue. I often miss this data in the Logs like here in the flight before, too.
https://drive.google.com/open?id=1so79TLIwRKzBpukxq8LTVFbIwB0EWD5M
QASSIST_SPEED=0, QASSIST_ANGLE = 30

IamPete1, 2020-05-14:
looks to me that a something caused a large error in roll causing a yaw error these both saturated the throttle so it gave up on yaw and prioritized roll and pitch until it was back on target again. For the whole 5 seconds or so its on full throttle until it get it back under control.

kd0aij, 2020-05-21:
I’m still looking for bugs in the tailsitter elevon controls. I did observe them both stuck at max up during a portion of one run, but unfortunately, I can’t find it in the logs. I believe I was running 4.0.5 at the time, but due to a problem with SITL in 4.0.5, I’ll continue the bugsearch in master.

IamPete1, 2021-08-28:
It is expected that BLHeli escs fix the min and max as it is not configurable in DSHot mode. Calibrations should not do anything when using DShot.

BLHeli is now used quite commonly, so I think I'm going to close, please re-open if I have missed something.

closed 2021-08-28 by IamPete1