ArduPilot/ardupilot #14294: TERRAIN directory failing to be created in the microSD

AngelAlbeniz, 2020-05-05:
**Issue details**

In short:
I started experiencing problems, related to the loading of terrain data. In the end I realized that the problem was that A.C 4.0.3 failed to create the microSD card directories correctly, when a clean, newly formatted microSD card was inserted into the Cube. The "TERRAIN" directory was always missing, making impossible for the terrain data to be loaded.

A more detailed explanation of all this incidence is in this post of Ardupilot Disclosure:
[https://discuss.ardupilot.org/t/unable-to-arm-prearm-waiting-for-terrain-data/54967](url)

I think it's better to read that explanation, I didn't want to make a big copy paste here.

**Version**
A.C 4.0.3

**Platform**
[  ] All
[  ] AntennaTracker
[X] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
Quadcopter

**Hardware type**
Cube Black, PowerBrick Mini, Here GPS, RFDx.......

**Logs**
I don't think they're relevant to this bug

AngelAlbeniz, 2020-05-25:
I'm seeing on the Ardupilot forum more cases of exactly this same problem.

[https://discuss.ardupilot.org/t/terrain-follow-problem/56872](url)
[https://discuss.ardupilot.org/t/terrain-not-working-on-cubeblack/56643](url)

It may be worth checking it out.

Hasi123, 2020-05-26:
I have the same issue with Cube Black, Here2 GNSS, RFDx

It is also not possible to create a folder with mavftp...