ArduPilot/ardupilot #14301: Diatone Mamba F405 MK2 Support

tenten8401, 2020-05-05:
## Feature request
It'd be cool to have ArduPilot support for the [Diatone Mamba F405 MK2 flight controller](https://pyrodrone.com/products/mamba-f405-flight-controller-mk2). It's inexpensive at only $20, has an integrated analog OSD, 4 motor outputs, 3 uarts, 3.3v, 5v & 9v BEC, i2c for compass and barometer and 16M flash attached. Betaflight has a target available: https://github.com/betaflight/unified-targets/blob/master/configs/default/DIAT-FURYF4OSD.config

**Platform**
[X] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

[image]
[image]

andyp1per, 2020-05-06:
It's pretty easy to construct firmware for a new board - all you need is an appropriate hwdef.dat and if you have the blueprint from betaflight then you are set. I suggest you try this. Look in libraries/AP_HAL_ChibiOS/hwdef and https://ardupilot.org/dev/docs/porting.html for examples and guidance.

tenten8401, 2020-05-06:
Thanks for the pointers, I'll have a go at it

tenten8401, 2020-05-07:
I've gotten pretty close to what I think is a working hwdef.dat, however I'm running into some weird RX DMA error that I'm not sure how to solve and can't find documented anywhere, nothing in that fmuv3 file either that'd point me in the right direction.. This is my hwdef.dat, most copied from MatekF405 and then pins changed according to the Pyro F4 target in betaflight. Here's my hwdef.dat and here's the error I'm getting: https://gist.github.com/tenten8401/7fc694c9014bfef4fdce2eeea05d6ac8
```[538/607] Compiling ArduCopter/compassmot.cpp
In file included from ../../libraries/AP_HAL/board/chibios.h:3:0,
                 from ../../libraries/AP_HAL/AP_HAL_Boards.h:130,
                 from ../../libraries/AP_HAL/AP_HAL.h:6,
                 from ../../libraries/AP_HAL_ChibiOS/UARTDriver.cpp:17:
./hwdef.h:336:65: error: 'STM32_UART2_RX_DMA_CONFIG' was not declared in this scope
 #define HAL_UART2_CONFIG { (BaseSequentialStream*) &SD2, false, STM32_UART2_RX_DMA_CONFIG, STM32_UART2_TX_DMA_CONFIG, 0, 0, 0, -1, 0, -1, 0}
                                                                 ^
./hwdef.h:339:82: note: in expansion of macro 'HAL_UART2_CONFIG'
 #define HAL_UART_DEVICE_LIST HAL_OTG1_CONFIG,HAL_USART3_CONFIG,HAL_USART1_CONFIG,HAL_UART2_CONFIG,HAL_UART4_CONFIG,HAL_UART5_CONFIG
                                                                                  ^~~~~~~~~~~~~~~~
../../libraries/AP_HAL_ChibiOS/UARTDriver.cpp:41:59: note: in expansion of macro 'HAL_UART_DEVICE_LIST'
 const UARTDriver::SerialDef UARTDriver::_serial_tab[] = { HAL_UART_DEVICE_LIST };
                                                           ^~~~~~~~~~~~~~~~~~~~
compilation terminated due to -Wfatal-errors.

../../libraries/AP_InertialSensor/AP_InertialSensor.cpp: In member function 'void AP_InertialSensor::detect_backends()':
./hwdef.h:139:110: error: 'ROTATION_YAW_0' was not declared in this scope
 #define HAL_INS_PROBE1  ADD_BACKEND(AP_InertialSensor_Invensense::probe(*this,hal.spi->get_device("mpu6000"),ROTATION_YAW_0))
                                                                                                              ^
../../libraries/AP_InertialSensor/AP_InertialSensor.cpp:786:61: note: in definition of macro 'ADD_BACKEND'
         if (((1U<<probe_count)&enable_mask) && _add_backend(x)) { \
                                                             ^
./hwdef.h:140:28: note: in expansion of macro 'HAL_INS_PROBE1'
 #define HAL_INS_PROBE_LIST HAL_INS_PROBE1
                            ^~~~~~~~~~~~~~
../../libraries/AP_InertialSensor/AP_InertialSensor.cpp:801:5: note: in expansion of macro 'HAL_INS_PROBE_LIST'
     HAL_INS_PROBE_LIST;
     ^
compilation terminated due to -Wfatal-errors.

Waf: Leaving directory `/home/isaac/ardupilot/build/PyroF4OSD'
Build failed
 -> task in 'objs/AP_HAL_ChibiOS' failed (exit status 1): 
	{task 140264545255712: cxx UARTDriver.cpp -> UARTDriver.cpp.0.o}
 (run with -v to display more information)
 -> task in 'objs/AP_InertialSensor/ArduCopter' failed (exit status 1): 
	{task 140264546378352: cxx AP_InertialSensor.cpp -> AP_InertialSensor.cpp.3.o}
```

Am I missing something obvious? Really new to this whole thing

andyp1per, 2020-05-07:
How are you building? Make sure you re-run ```./waf configure --board Mamba405``` (or whatever you call it) and ```./waf clean``` each time

hwdef.h is generated from hwdef.dat during these steps, it looks like you are missing some generated defines

tenten8401, 2020-05-07:
@andyp1per That's exactly what I'm running: https://gfycat.com/organicrealisticdassie
It doesn't happen on the other boards in the same projects so I'm not sure really where to go from here..

andyp1per, 2020-05-07:
Can you create a draft PR with your changes so that I can look?

tenten8401, 2020-05-07:
Sure

tenten8401, 2020-05-07:
@andyp1per https://github.com/ArduPilot/ardupilot/pull/14313

jeanphilippehell, 2020-05-09:
Hi ! I am interested and currently trying to get the Mamba F405 MK2 running Ardupilot.
I managed to get the board running without baro, see https://discuss.ardupilot.org/t/disable-barometer/55868
Now I'm trying to get the SBUS input

tenten8401, 2020-05-10:
@jeanphilippehell I see you're just flashing the OmnibusF4 target, I doubt that will work real well. I have the Mamba F405 coming in a few days and will create a target for it when it arrives.

jeanphilippehell, 2020-05-10:
@tenten8401 Yes as I am yet not used to create new targets for Ardupilot so I just wanted to experiment things.
But I really would like to use my Mamba F405 in the next few days so I will continue to dig around, and once your PR is completed I will use the new PyroF4OSD/Mamba F405 target.
In the meantime I could help you by providing my results or try yours if you want.
JP

tenten8401, 2020-05-10:
@jeanphilippehell I'll see if I can create some sample F405 hex files if you're willing to test them for me. Give me a little bit

jeanphilippehell, 2020-05-10:
@tenten8401 no worries, tell me when they are ready and I will test them

tenten8401, 2020-05-10:
@jeanphilippehell Can you try these hex files out? SERIAL1, 3 and 6 should map to the corresponding pins on the board.. not sure if the other ones will be there or not or what they do.. My board will be here Tuesday to test this out on
[MambaF405.zip](https://github.com/ArduPilot/ardupilot/files/4606563/MambaF405.zip)

jeanphilippehell, 2020-05-10:
@tenten8401 sorry but I'm more used to flash the .bin files. Could you send me those please ?

tenten8401, 2020-05-10:
@jeanphilippehell Here you go
[ardu bin.zip](https://github.com/ArduPilot/ardupilot/files/4606626/ardu.bin.zip)

jeanphilippehell, 2020-05-10:
thanks ! In the meantime I learnt to flash the .hex files with the Betaflight Configurator. Sorry about that ...
Anyway, when I flash any of the .bin files the board does not start nor appears on USB.
When flashing the .hex with Betaflight Configurator the board starts.
On QGroundControl the communication, MAVLink parameters and Horizon works with the .hex firmwares.
The RC SBUS in does not work.
I need to prepare the wiring to test the Serial 3 and 6

tenten8401, 2020-05-10:
Is the SBUS just RX1? Have you set SERIAL1_FUNCTION to RCIN?

tenten8401, 2020-05-12:
So it just occurred to me that the Mamba F405 Mini MK2 is the same board just shrunken to a 20x20 form factor, have one of those on the way too so it'll be getting tested as well.
@jeanphilippehell Did you ever get SBUS working?

jeanphilippehell, 2020-05-12:
Hi @tenten8401 ,
I have tried to set SERIAL1_FUNCTION to None, Frsky SBUS and with or without RX Invert, with no result.
The IMU need to to be rotated to Yaw90.
I get an error on QGC about a missing parameter, COMPASS_PRIMARY, but that seems normal since there is no compass on this board.
The OSD is fried on my board so I won't be able to test that.
On my side I made some progress, I managed to get the SBUS working as well as the RC Outs 1-4.
Basically I had to change the Timers used by the RC Out pin so they doesn't use the same timer as PA10 / RCININT
Tonight I will push my repo so you can compare with your work
JP

tenten8401, 2020-05-12:
Alright, I'll keep an eye out. Thanks.

tenten8401, 2020-05-14:
Seems my shipment is "Running late" with no ETA so it may be a while before I can test anything.

jeanphilippehell, 2020-05-14:
Hi, sorry to hear that.
I've started to push my new branch "MambaF405" on my repo :
https://github.com/jeanphilippehell/ardupilot/commits/MambaF405
Next I will solder some wires on my board to check if the serial ports works.
While waiting your shipment I can continue to test your binaries of course
JP

tenten8401, 2020-05-16:
So for some reason on my board I can't initialize with this error:
[image]
Not even sure where to go from here as I'm just copying everything from betaflight FURYF4OSD and it works fine there.. It's on the right spi bus & all that as far as I can tell.

andyp1per, 2020-05-17:
It's using external interrupts in BF for the MPU6000 (other targets don't seem to). I don't know how you specify that in ardupilot but I suspect it's not what you have. Also shouldn't the IMU definition come after the SPI selection?

vierfuffzig, 2020-05-17:
see https://github.com/ArduPilot/ardupilot/blob/05fe49a51f36bcd0ed0e9dd09aa5776b479a9dad/libraries/AP_HAL_ChibiOS/hwdef/revo-mini/hwdef.dat#L79
for reference. not sure if that solves the issue, didn‘t spot anything else at rough glance though.

tenten8401, 2020-05-17:
@andyp1per You were correct, it was the IMU definition order. Derp.
Seems all good now, looks like it's just small tweaks from here and making sure all the uarts work and whatnot..

tenten8401, 2020-05-18:
Uarts seem to work as far as I can tell, GPS is communicating properly. Tomorrow I'll test I2C and motor outputs. I'm not sure what the status on WS2812B led support is but I could also give that a try

andyp1per, 2020-05-18:
It would be worth trying the external interrupts setting that @vierfuffzig gave above

callumMacdermid, 2020-05-24:
Fairly new to Ardupilot but I have a Diatone F405 that I'd be happy to help to test with!

tenten8401, 2020-05-25:
@callumMacdermid Would you mind testing out I2C baro or compass and making sure all the uarts work here? Should be SERIAL1-3.
[bin.zip](https://github.com/ArduPilot/ardupilot/files/4674969/bin.zip)

jeanphilippehell, 2020-05-26:
Hi ! A quick update on my side : tonight I got the Dataflash logger working.
As for now I have the following working :
- IMU
- Sbus input
- RC out 1-4
- Battery voltage
- Dataflash
I've pushed my latest hwdef.dat here :
https://github.com/jeanphilippehell/ardupilot/blob/MambaF405/libraries/AP_HAL_ChibiOS/hwdef/MambaF405/hwdef.dat
Hope it helps. I am flying the Mamba F405 on a fixed wing aircraft with no GPS, telemetry, Baro nor Compass.

tenten8401, 2020-05-27:
I'll get to testing the GPS & baro on my end and if I run into anything weird I'll try taking some of the changes you've done unless you'd rather work on a PR

Sekilsgs2, 2020-09-14:
What about working arfu for this board?

andyp1per, 2020-09-14:
@jeanphilippehell @tenten8401 are you able to do a PR for this so that we can get suport in?

Sekilsgs2, 2020-09-14:
> @jeanphilippehell @tenten8401 are you able to do a PR for this so that we can get suport in?

If you little helping for me - im can try make pr. 
What im need - sbus not working - im try edit hwdef - add uart1 as simple uart - in mission planer im change serial1 protocol to 23 (rcin) - not working..
Im try add rx pin from uart1 as rcin as in @jeanphilippehell - than change serial1 proto to 23 - not working - not detecting..
Im try add pc0 gpio with low value - in this pcb this hw inverter - not helping.. what wrong ? :)

jeanphilippehell, 2020-09-15:
Hi @Sekilsgs2 , thanks for your interest

This is my configuration with sbus :
- R-XSR with the SBUS_OUT/CPPM wire soldered to the SBUS pad of the MambaF405 (not the PPM!)
- **No** SERIAL parameter set for RC_IN, Sbus etc
- There is **no** parameter to set : the RC_IN is set via my [hwdef.dat](https://github.com/jeanphilippehell/ardupilot/blob/MambaF405/libraries/AP_HAL_ChibiOS/hwdef/MambaF405/hwdef.dat) see lines 80 and 81
- Ardupilot should see the RC signals without having to set up anything.

I've removed the irrelevant and misleading comment about PPM/SBUS solder pad etc, it was from another board from which I first copied the hwdef.dat. Sorry about that

Note that my work is based on Plane 4.0.5 more specifically this SHA1 : 0bfa2638ce499b2b515df73c6d389b22a64f61e7

I will see if it is easy to rebase my work on master an then make a PR

jeanphilippehell, 2020-09-15:
Ok I skipped the part about using the[qczek]( http://qczek.beyondrc.com/qczek-lrs-433mhz-1w-lora-rc-link/) receiver.
I don't know this (interesting) receiver, but if it works with Inav on the UART1/S.BUS pad then it should work with this hwdef.dat

IIRC there is no way to invert uart signals via software on the STM32F4 (only from F7/H7 if I'm right)

tenten8401, 2020-09-15:
Yeah I'll see if I can get a PR made soon, the rig I was using to test the mamba board has another board in it now so I need to just solder it all up and test it all.

@jeanphilippehell RCIN should autoinvert since it's softserial but no hardware uart iirc

jeanphilippehell, 2020-09-15:
@tenten8401  thanks for the input !

I can make the PR, it will be my first PR : a good exercise I think

Sekilsgs2, 2020-09-16:
Im try as from your hwdef - uart1 as rcin - no signal from qczek
Only when im add ppm pin as rcin:                                           PC9 TIM8_CH4 TIM8 RCININT PULLDOWN
With this im get signal from ppm pin.. 
This board have hw inverter PC0

jeanphilippehell, 2020-09-16:
Hi @Sekilsgs2 ,

I've attached the .apj file I used for my board, along with the .param file to upload via QGroundcontrol.
With these files you should be in the same configuration than me.
They are based on Arduplane 4.0.5 but they will allow you to test your HW configuration with the qczek receiver.

I've just tested the switch between SBUS and CPPM with my R-XSR and Ardupilot is able to parse the RC signals in both mode.
The SBUS ouput from the R-XSR is supposed to be inverted, based on this site :
[https://oscarliang.com/uninverted-sbus-smart-port-frsky-receivers](https://oscarliang.com/uninverted-sbus-smart-port-frsky-receivers)
However I haven't found confirmation or further information in the [manual](https://www.frsky-rc.com/wp-content/uploads/Downloads/Manual/R-XSR/R-XSR%20ACCST%20-Manual.pdf)
Thus I am guessing that the R-XSR Firmware is sending normal SBus signals that are electronically inverted when in SBus mode (blue led on).
When in CPPM mode (blue led off) I don't know if the R-XSR sends normal CPPM (which would require that the pulses from the µc are inverted then electronically re-inverted), or the CPPM output from the R-XSR is inverted which would be weird.
The R-XSR is connected to the SBUS pad of the MambaF405.
There is a little component nearby on the board that could be a transistor to invert the signal.
So I would say that the SBUS pad of the MambaF405 does have an inverted input.

Let me know if you manage to read the RC signals with qczek wired on the SBUS pad of the MambaF405 with the .apj and .param attached in the .ZIP archive.

JP
[MambaF405_Arduplane4_0_5.zip](https://github.com/ArduPilot/ardupilot/files/5234738/MambaF405_Arduplane4_0_5.zip)

Sekilsgs2, 2020-09-16:
Ok thank - im try but tomorrow - im reply..

Sekilsgs2, 2020-09-20:
Working sbus - this my mistake. I find that timer channel for motor same as uart 1 rx pin - im change to other for motor and now all working i2c uarts sbus.. 
Make good pr for this board or im can make this..

jeanphilippehell, 2020-09-21:
Hi, I've managed to polish my work and start my PR :

https://github.com/ArduPilot/ardupilot/pull/15392

closed 2021-07-11 by tenten8401