ArduPilot/ardupilot #14305: EKF3 Plane reports emergency yaw reset and stops logging

Hwurzburg, 2020-05-06:
Been flying my omnibusf4pro with master based firmware that builds only EKF2 to avoid the EKF3 issues with micro GPS....but just did some test flights today with master and EKF3, changing the EKF3_CHECK_SCALE to 120 to allow it to arm quickly.....multiple flights went flawlessly, but when I checked the logs to find my recorded current consumption and cruise currents,  I found that both flight logs had been truncated after the AUTO takeoff, right after an EKF3 emergency yaw reset at the end of the AUTO mission to takeoff....strangely stating that the mag stopped....but there is no mag and no COMPASS_USEx enabled!
It instantly resumes  using GPS and then the log stops....these were 10 and 30 minute flights

logs: https://www.dropbox.com/s/d715jmxv55og7it/00000001.BIN?dl=0
https://www.dropbox.com/s/3u4hnjcom6ys352/00000002.BIN?dl=0
**Platform**
[  ] All
[  ] AntennaTracker
[  ] Copter
[X  ] Plane
[  ] Rover
[  ] Submarine

vierfuffzig, 2020-05-06:
afaik as for now you need to set EK3_MAG_CAL=7 (GSF etimator) for inflight GPS yaw alignment on non-mag setups using EK3. probably there's some room for improvement here...

peterbarker, 2020-05-07:
On Wed, 6 May 2020, Henry Wurzburg wrote:

> EKF3 emergency yaw reset at the end of the AUTO mission to takeoff....strangely stating that the mag stopped....but there is no mag and no COMPASS_USEx

That error message is misleading.  It's actually trying to tell you it is 
about to disable the magnetometers.  We have removed reference to 
magnetometers in that message in master.

> It instantly resumes using GPS and then the log stops....these were 10 and 30 minute flights

Could try another SD card, please?  Just set it logging on the bench and 
wait, I suggest.  I *hope* the not-logging and resets are unrelated...

Peter

Hwurzburg, 2020-05-07:
this was master-7 commits as of this morning

Hwurzburg, 2020-05-07:
I think its coming from here: https://github.com/ArduPilot/ardupilot/blob/0c1ba48212eac9cb871bfdf7993b4fe2e0fa01c3/libraries/AP_NavEKF3/AP_NavEKF3_MagFusion.cpp#L1504

Hwurzburg, 2020-05-07:
apparently it got changed from a use_compass test to  a MAG_CAL = GSF test ....beyond my capabilities as to why....but it appears that to not use compass in plane now requires switching as Basti suggests, but that means all the GSF and IMU masks also must be correct also....definitely different than EKF2

Hwurzburg, 2020-05-07:
Peter, indeed it was the SD card....it was an older one (4GB)..why it was working on old builds with EKF2, and stopped on the latest is a mystery....must be a subtle timing thing...so all is left is to decide what to do about the non compass case of plane

priseborough, 2020-05-08:
@Hwurzburg In the short term you can set EK3_MAG_CAL = 7 to let the EKF know that non use of compass is intended by the user. I am currently going through thecompass library methods to see which can be used by the EKF to reliably indicate that the non use of compass for yaw was intended by the user - @tridge FYI

priseborough, 2020-05-08:
I was thinking originally of using Compass::use_for_yaw(index) function to enable the EKF3 to determine here https://github.com/ArduPilot/ardupilot/blob/9eaf7f552820e5bcd653605f6c670cf960bac1e9/libraries/AP_NavEKF3/AP_NavEKF3_MagFusion.cpp#L1501-L1505 if the user intended compass use or not, but it is not just a function of  the COMPASS_USEx parameters as it will also return false if inflight compass learning is running.

tridge, 2020-05-08:
I think we should add a num_enabled_sensors() in AP_Compass, giving the number of discovered sensors that have COMPASS_USE set

tridge, 2020-05-08:
fixed by #14323

PR #14323 (2020-05-09): Fixed scary message in EK3 on yaw reset from GSF when no compass
fixes #14305
closed 2020-05-08 by tridge